YuuichiAkagawa/USBH_MIDI #19: Please help me!

MGR10, 2015-02-20:
I have Leonardo and nanopad2.
I´m not able to run midi out.
What I´m doing wrong?
Must I write some think in code or only load library ?
Thank you very much.

YuuichiAkagawa, 2015-02-20:
@MGR10 
I have same issue on nanoPAD.
nanoPAD needs special driver.　It does not work with generic USB MIDI driver.

MGR10, 2015-02-21:
Thank you for answer.If I'll use nanokontrol must I change somethink in
code?Or simple upload sketch to Arduino leonardo.?
Sorry I 'm beginner with arduino.

>
> @MGR10
> I have same issue on nanoPAD.
> nanoPAD needs special driver. It does not work with generic USB MIDI
driver.

spiderWAW, 2015-02-21:
Nanokontrol 2 works without problems. But should be set to CC mode.

MGR10, 2015-02-21:
Great!
I prefer CC because I built footcontroller for Kemper amp and Ableton looper.
So what I need to change in code?
Only SERIAL1?

spiderWAW, 2015-02-21:
From my POV nothing… Boot nanokontrol in this mode > read manual/net.

MGR10, 2015-02-25:
I guess I'm not stupid ,but I uploaded this cobe in to the leonardo.
and nanokontrolor doesn't want to boot.

Please !help !!!!

.......................................................................................................................................................
USB  Usb;
USBH_MIDI  Midi(&Usb);

void MIDI_poll();
void doDelay(unsigned long t1, unsigned long t2, unsigned long delayTime);

void setup()
{
  Serial1.begin(31250);

  //Workaround for non UHS2.0 Shield 
  pinMode(7,OUTPUT);
  digitalWrite(7,HIGH);

  if (Usb.Init() == -1) {
    while(1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  unsigned long t1;

  Usb.Task();
  t1 = micros();
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {
    MIDI_poll();
  }
  //delay(1ms)
  doDelay(t1, micros(), 1000);
}

// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
    byte outBuf[ 3 ];
    uint8_t size;

    do {
      if( (size=Midi.RecvData(outBuf)) > 0 ){
        //MIDI Output
        Serial1.write(outBuf, size);
      }
    }while(size>0);
}

// Delay time (max 16383 us)
void doDelay(unsigned long t1, unsigned long t2, unsigned long delayTime)
{
    unsigned long t3;

    if( t1 > t2 ){
      t3 = (4294967295 - t1 + t2);
    }else{
      t3 = t2 - t1;
    }

    if( t3 < delayTime ){
      delayMicroseconds(delayTime - t3);
    }
}

YuuichiAkagawa, 2015-02-26:
@MGR10 
I test your sketch, no problem found. It work fine.

My environment is
* Leonardo R3 + USB Host Sheild 2.0 + Sparkfun MIDI Brakeout
* nanoKONTROL
* Arduino IDE 1.0.6 + Latest USB Host Library Rev.2.0 + Latest USBH_MIDI library

[image]

MGR10, 2015-02-26:
Thank you,reply.
I using without Usb shield !?Is it problém.?

YuuichiAkagawa, 2015-02-26:
USB Host Shield is required.
Where did you connect the MIDI device ?

MGR10, 2015-02-26:
There is problem, If I plug nanokontrol in microusb Leonardo, Nanokontrol
only blink up every leds and then nothing.

MGR10, 2015-02-26:
Ok.
Tomorow I received USB Host Shield.
Thank you for now.
MG

MGR10, 2015-02-27:
Hi[?]

I received usb host shield  and still nothing

Nanocontrol don=t boot.

Any suggestion

Thak you

2015-02-26 17:26 GMT+01:00 Mirek Gregor <gregormirek10@gmail.com>:

MGR10, 2015-02-27:
Ok
Solved. Two bad usb cabel.
Thak you

closed 2015-02-27 by YuuichiAkagawa

MGR10, 2015-02-28:
Hello again
I tried Nanopad2 end works like charm.,too.