YuuichiAkagawa/USBH_MIDI #20: CC to Sysex translator

MGR10, 2015-03-08:
hello again.

How can I translate incoming data/ CC / from nanopad  in to Sysex and sent via midi out.
Or If I know Sysex code and I need send them out.

Thank you
Mirek

YuuichiAkagawa, 2015-03-09:
You can use with Arduino MIDI Library.
https://github.com/FortySevenEffects/arduino_midi_library

CC handling sample 
```C++
void MIDI_poll()
{
    byte outBuf[ 3 ];
    uint8_t size;
    do {
      if( (size=Midi.RecvData(outBuf)) > 0 ){
        if((outBuf[0] & 0xf0) == 0xb0){ //CC received?
          //Send sysex to serial code here
        }
      }
    }while(size>0);
}
```

Please write your code in '//Send sysex to serial code here' line

MGR10, 2015-03-09:
I appreciate your help, but we have to a lot to teach, to understand how to write it.
I am absolute beginner.
As I wrote,I m guitar player and I made Diy footcontroler with nanokonrol and arduino.
I have Kemper amp , and it have looper inside,but only for NRPN data,no CC.
I know NRPN , if I send you,Would you please write me code.
Thank you very very much.Mr.

YuuichiAkagawa, 2015-03-10:
Hmm...
You should recruit tech member.

MGR10, 2015-03-10:
Hmm..
Ok. Its no commercial controler, but my handmatě Diy.
Doesn't matter,I need only send NRPN data.
I go to study arduino codes,hoe write it.
I appreciate your answers.
MG

MGR10, 2015-03-10:
Ha .
I found code for Jazz plugin.
I must admit I am in ....

Here is only fragment about midi_ not html code:
//// Callback function for MIDI in
function midiProc(t,a,b,c){
	
	if(a>0xaf)
	{
		if(a==0xb0 && b==MIDI_CC_TO_LISTEN) // b = CC#
		{
			document.getElementById("input").innerHTML = "MIDI input CC#/value: "+ b +" / "+ c;
			if(c == prev_cc_val) same_cc_cnt++;
			else same_cc_cnt = 0; 
			prev_cc_val = c;
	
			
			switch(c) { // c = CC value
			    case 0:
			        // start loop record
					document.getElementById("command").innerHTML = "Last sent Looper command: "+"<b><span style=\"color:blue\">RECORD / PLAY / OVERDUB</b></span>";
					
					switch(looper_state){
						case "EMPTY":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01]); // just start recording						
							looper_state = "RECORD";
						break;
						case "STOP":
							// issue play
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x00]);
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x01]);
							looper_state = "PLAY";
						break;
						case "PLAY":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01]); // just start recording
							looper_state = "RECORD";
						break;
						case "RECORD":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x00]); // just stop recording & play						
							looper_state = "PLAY";
						break;
					}
			        break;
			    case 1:
					// stop / 3x for erase
					document.getElementById("command").innerHTML = "Last sent Looper command: "+"<b><span style=\"color:blue\">STOP / 3x to DELETE</b></span>";
					Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5E,0xB0,0x06,0x00,0xB0,0x26,0x01]);
					
					switch(looper_state){
						case "EMPTY":

						break;
						default:
							if(same_cc_cnt == 2)
							{
								prev_cc_val = -1; 
								same_cc_cnt == 0;
								looper_state = "EMPTY";
							}
							else looper_state = "STOP";
						break;
					}
						break;

					break;
			    case 2:
			        //  undo / redo
					document.getElementById("command").innerHTML = "Last sent Looper command: "+"<b><span style=\"color:blue\">UNDO / REDO</b></span>";
					Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x01]);
					break;
			    case 3:
					// trigger
					document.getElementById("command").innerHTML = "Last sent Looper command: "+"<b><span style=\"color:blue\">TRIGGER</b></span>";
					Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5A,0xB0,0x06,0x00,0xB0,0x26,0x01]);
			        break;
			    case 4:
			        // reverse
					document.getElementById("command").innerHTML = "Last sent Looper command: "+"<b><span style=\"color:blue\">REVERSE</b></span>";
					switch(looper_state){
						case "PLAY":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5B,0xB0,0x06,0x00,0xB0,0x26,0x01]);
						break;
						case "STOP":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5B,0xB0,0x06,0x00,0xB0,0x26,0x01]);
							// issue play
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x00]);
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x01]);
							looper_state = "PLAY";
						break;
					}
			        break;
			    case 5:

			        // 1/2 time
					document.getElementById("command").innerHTML = "Last Looper command sent: "+"<b><span style=\"color:blue\">1/2 TIME</b></span>";
					switch(looper_state){
						case "PLAY":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5C,0xB0,0x06,0x00,0xB0,0x26,0x01]);
						break;
						case "STOP":
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5C,0xB0,0x06,0x00,0xB0,0x26,0x01]);
							// issue play
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x00]);
							Jazz.MidiOutLong([0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x01]);
							looper_state = "PLAY";

MGR10, 2015-03-10:
Here are only NRPN data

START RECORDING:

0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01

STOP RECORDING AND PLAY:

0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x00

REVERSE: 

0xB0,0x63,0x7D,0xB0,0x62,0x5B,0xB0,0x06,0x00,0xB0,0x26,0x01

1/2 SPEED:

0xB0,0x63,0x7D,0xB0,0x62,0x5C,0xB0,0x06,0x00,0xB0,0x26,0x01

UNDO:

0xB0,0x63,0x7D,0xB0,0x62,0x5D,0xB0,0x06,0x00,0xB0,0x26,0x01

STOP:

0xB0,0x63,0x7D,0xB0,0x62,0x5E,0xB0,0x06,0x00,0xB0,0x26,0x01

ERASE:

0xB0,0x63,0x7D,0xB0,0x62,0x5E,0xB0,0x06,0x00,0xB0,0x26,0x00

MGR10, 2015-03-11:
Hello

I try write code.
But many bugs.

[code]
// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
    byte outBuf[ 3 ];
    uint8_t size;

    do {
      if( (size=Midi.RecvData(outBuf)) > 0 ){
        if((outBuf[0] & 0xf0) ==0xb0,0x01,0x00){ //CC received? CC#
        Serial1.write(0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01);//Send sysex to serial code here
        Serial1.write(outBuf, size);
      }
      }
    }while(size>0);
}
[code]
 Thanks

YuuichiAkagawa, 2015-03-11:
Fix like this.
```
const byte nrpnStartRecording[] = {0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01};
// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
    byte outBuf[ 3 ];
    uint8_t size;
    do {
      if( (size=Midi.RecvData(outBuf)) > 0 ){
        if((outBuf[0] & 0xf0) == 0xb0){ //CC received?
          //Send sysex to serial code here
          if(outBuf[1] == 0x01 && outBuf[2] == 0x00){
            Serial1.write(nrpnStartRecording, sizeof(nrpnStartRecording));
          }
        }
      }
    }while(size>0);
}
```

MGR10, 2015-03-11:
Thank you very much.
MG

MGR10, 2015-03-12:
const byte nrpnStartRecording[] ={0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01};
 // Poll USB MIDI Controler and send to serial MIDI
 void MIDI_poll()
 {
     byte outBuf[ 3 ];
     uint8_t size;

     do {
       if( (size=Midi.RecvData(outBuf)) > 0 ){
         if((outBuf[0] & 0xf0) ==(0xb0,0x50,0x7f));{ //CC received?
          if(outBuf[1] == 0x01 && outBuf[2] == 0x00){
             Serial1.write(nrpnStartRecording, sizeof(nrpnStartRecording));
       }
     }while(size>0);

     return;
 }

 // Delay time (max 16383 us)
 void doDelay(unsigned long t1, unsigned long t2, unsigned long delayTime)
 {
     unsigned long t3;

     if( t1 > t2 ){
       t3 = (4294967295 - t1 + t2);
     }else{
       t3 = t2 - t1;
     }

     if( t3 < delayTime ){
       delayMicroseconds(delayTime - t3);
     }
 }

closed 2015-03-12 by MGR10

MGR10, 2015-03-12:
今日は

I wrote code but error masage
-------------------------------------------------------------------------------------------------------------
const byte nrpnStartRecording[] = {0xB0,0x63,0x7D,0xB0,0x62,0x58,0xB0,0x06,0x00,0xB0,0x26,0x01};
// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
    byte outBuf[ 3 ];
    uint8_t size;

    do {
      if( (size=Midi.RecvData(outBuf)) > 0 ){
        if((outBuf[0] & 0xf0) ==(0xb0,0x50,0x7f)){ //CC received?
         if(outBuf[1] == 0x01 && outBuf[2] == 0x00){
            Serial1.write(nrpnStartRecording, sizeof(nrpnStartRecording));
      }
    }while(size>0);
    
    return;
}
-------------------------------------------------------------------------------------------------------

 
USB_MIDI_converter.ino: In function 'void MIDI_poll()':
USB_MIDI_converter:76: error: a function-definition is not allowed here before '{' token
USB_MIDI_converter:88: error: expected `}' at end of input
USB_MIDI_converter:88: error: expected `while' at end of input
USB_MIDI_converter:88: error: expected `(' at end of input
USB_MIDI_converter:88: error: expected primary-expression at end of input
USB_MIDI_converter:88: error: expected `)' at end of input
USB_MIDI_converter:88: error: expected `;' at end of input
USB_MIDI_converter:88: error: expected `}' at end of input

YuuichiAkagawa, 2015-03-12:
OMG, please learning programming.
I am not your teacher.

MGR10, 2015-03-12:
Thank you
Goodbye

MGR10, 2015-03-12:
I solved errors in code, but your code doesn't work.
You can cloded it