YuuichiAkagawa/USBH_MIDI #21: Vendor ID

benbiles, 2015-07-08:
Hi, The example code has detected my midi device Vendor ID / Product ID.
Do I need to edit this into the code somewhere? its not clear where if I do ?
other people asking the same thing, http://forum.arduino.cc/index.php?topic=277315.0
or are the detected values are used?
Also in dump example shoudl the midi values be dumped to the screen as keys are pressed / knobs turned etc ?I just get a load of numbers repeated scrolling and no change as keys are pressed..

I am using USB HOST LIB version 2

YuuichiAkagawa, 2015-07-08:
Vendor ID and Product ID is not used. It is just information.

This library use an USB device class information.
It is Implemented in USBH_MIDI::parseConfigDescr().
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/usbh_midi.cpp#L262-307

Please tell me USB descriptor.
You can use USB Host Shield 2.0 Library's example 'USB_desc'.

benbiles, 2015-07-08:
Hi, I ran USBdesc and got:

Start
01
--
Request error. Error code:	0D
Addr:1(0.0.1)

In USBH_midi dump I get:
VID:1235, PID:0034

I think the device uses midi channel 19 for control knobs and I am trying to use the byte data from your library to control things from the arduino Arduino.

YuuichiAkagawa, 2015-07-08:
Hmm...
Your MIDI device is "Launch Control" ?
Do you have enough power ?

benbiles, 2015-07-08:
Just tried with 12v external supply on arduino and USB to mac, should be enough power..
same error when running USBdesc
I added #include <SPI.h> to you example code since arduino would not compile without,.
Is this wrong maybe ??

benbiles, 2015-07-08:
novation Launchcontrol http://global.novationmusic.com/launch/launch-control#

YuuichiAkagawa, 2015-07-08:
Sorry, I have no idea.
If  USB Host Shield 2.0 Library cannot handle the device, my library can't too.
(This library depend to USB Host Shield 2.0 Library.)

Cause of the SPI.h error is due to a minor change in the USB Host Shield 2.0 Library .
https://github.com/felis/USB_Host_Shield_2.0/commit/50b6ce4c9c7051e61bc69ae8bf411bd3256b8fa6
I will fix later. Thank you for your report.

benbiles, 2015-07-08:
Hi, yes agree, will try to find out why device is not handled by USBhost first. I had assumed your library
was working since Hex numbers were streaming into the console after VID PID reported.

I found this code that reports Novostation launchpad working with USBhost2 so I will see if there is
a hack here that will make the lanchcontrol detect in USBhist. https://github.com/j0uni/LaunchpadArduino

benbiles, 2015-07-09:
Hi Yuuichi, Just to say you were correct. It was lack of power even though the Arduino Uno was plugged in with the 12v power adapter !! 

I used a USB hub with 
#include <usbhub.h>
USB  Usb;
USBHub Hub1(&Usb);

added to your USBH_MIDI_dump example code and now I get streams of Hex code from the Launch Control.

So I guess you can add the Novation Launchcontrol to your list of working midi devices!!

Now I'm printing basic number values from volume pots with

byte basicnumber = bufMidi[i];
Serial.print(basicnumber);

great !! thanks so much for this library, its really awsome !!

YuuichiAkagawa, 2015-07-09:
:smile: Enjoy !

closed 2015-07-09 by YuuichiAkagawa