YuuichiAkagawa/USBH_MIDI #23: Promlem with bidirectional and MIDI 4.2 library

spiderWAW, 2015-09-16:
bidrectional_converter.ino: In function 'void setup()':
bidrectional_converter:52: error: 'MIDI' was not declared in this scope
bidrectional_converter.ino: In function 'void loop()':
bidrectional_converter:74: error: 'MIDI' was not declared in this scope
bidrectional_converter.ino: In function 'void MIDI_poll()':
bidrectional_converter:98: error: 'MIDI_DEFAULT_SERIAL_PORT' was not declared in this scope

YuuichiAkagawa, 2015-09-17:
What are your platform?
OS, IDE Version, Arduino board, etc.

spiderWAW, 2015-09-17:
OS X 10.10, IDE 1.0.5, UNO r3. No problem with other sketches. On IDE 1.6.5 I have other problem: 

In file included from /Documents/Arduino/libraries/USB_Host_Shield_2/Usb.h:27:0,
                 from bidrectional_converter.ino:27:
/Documents/Arduino/libraries/USB_Host_Shield_2/settings.h:139:176: fatal error: SPI.h: No such file or directory
 #include <SPI.h> // Use the Arduino SPI library for the Arduino Due, RedBearLab nRF51822, Intel Galileo 1 & 2, Intel Edison or if the SPI library with transaction is available
                                                                                                                                                                                ^
compilation terminated.
Error compiling.

YuuichiAkagawa, 2015-09-17:
Hmm...

I tried OS X 10.10.5(Japanese) + IDE 1.0.5 + UNO R3.
Libraries install to $HOME/Documents/Arduino/libraries/

No compile error. 
IDE 1.6.5 too.

spiderWAW, 2015-09-17:
Very interesting...  problem with IDE 1.6.5 and missing libraries exist on Windows too (see here: http://www.vguitarforums.com/smf/index.php?topic=9681). When I use IDE 1.0.5 and newest your library all is ok. 

But even with 1.0.5 when I use MIDI library 4.0.2 I have error too (see first post).

YuuichiAkagawa, 2016-04-03:
@spiderWAW 
USBH_MIDI is now included in USB Host Shield 2.0 Library.
Please update tutorial document.

closed 2016-04-03 by YuuichiAkagawa