YuuichiAkagawa/USBH_MIDI #24: Device list

jackbite, 2015-10-01:
I'm currently testing both receive and send via usb using a roland td-11, alesis micron, and microkorg xl.  Receive works great but I cannot get send to work for any of them.  Any thoughts on where to start debugging?

benbiles, 2015-10-01:
Hi jackbite , did you try the mididump example , i used that to get the midi values from the novation controler. Also I found that the usb shield i was using did'nt have enough bus power to run the control surface so I added the USB shield usbhub fix and powered with a hub. don't know if any of this helps?

YuuichiAkagawa, 2015-10-01:
I don't provide debug tool for SEND.

MIDI channel is correct?

jackbite, 2015-10-01:
To benbiles: I can get the data dump info but would not know where to plug it into the 2.0 code.

To YuuichiAkagawa: MIDI channel is correct and the note on, note off values are correct according to the MIDI channel selected on the device.  Could I send you my code to review for any glaring mistakes?

YuuichiAkagawa, 2015-10-02:
@jackbite 
Please put your code on gist. 
And tell me it location.

jackbite, 2015-10-05:
<script src="https://gist.github.com/jackbite/9adf0bca75f894ed04a2.js"></script>

jackbite, 2015-10-05:
https://gist.github.com/9adf0bca75f894ed04a2.git

Sorry!

jackbite, 2015-10-05:
https://gist.github.com/jackbite/9adf0bca75f894ed04a2/archive/da287df7054f8d088fe7a9acc4513b9c100e162b.zip

Doh!!

YuuichiAkagawa, 2015-10-06:
@jackbite 
I don't understand this condition.
```C++
while ((micros() - recStartTime > recTime[recPointer]) && (f == 0)){
```
It is simply,
```C++
while (f == 0){
```
And you must fix the array size.
```C++
byte rec[3000+1];
unsigned long recTime[3000+1];
```
https://gist.github.com/YuuichiAkagawa/72ae1d33e0f76f341e43/revisions

I don't have Teensy board and your MIDI devices.
I test with Arduino UNO + Roland UME-ONE + QUAD-CAPTURE + PC
My devices has no problem.

Arduino --USB--> Roland UME-ONE  --MIDI--> QUAD-CAPTURE --USB--> PC(MIDI-OX)
*MIDI keyboard connect to UME-ONE's MIDI in

Arduino version sketch
https://gist.github.com/YuuichiAkagawa/58dfa6ad1fb7a979c241
* array size is shrink to 100, alternative interval timer

ykrkn, 2015-11-10:
Was runned with iRig Keys VID:1963, PID:0005
Tested using MIDI Monitor. All works as good as expected. Thank you for the great sources!

YuuichiAkagawa, 2015-11-10:
@ykrkn :smile:

closed 2016-04-02 by YuuichiAkagawa