YuuichiAkagawa/USBH_MIDI #25: Compatibility with Zero / M0 / Due native USB Host?

DisasterAreaDesigns, 2015-11-16:
Hi, the new version works great with the MAX3241 based USB Host Shield!  Arduino.cc have adapted Oleg's v2.0 library as part of the new USBHost library to run on the ARM / SAM based boards.  How difficult do you think it would be to port your library to run on the native USB host interface of these boards?

YuuichiAkagawa, 2015-11-16:
Hello, @DisasterAreaDesigns .

No. It is not compatible.
ARM based USBHost library API is slightly different.
Currently, there are no plans to support. If there is a lot of request, i will update library.

*I don't have these boards.

DisasterAreaDesigns, 2015-11-16:
I would be happy to send you an ARM board if you would consider adding support for the native USB host.

bbx10, 2015-12-21:
https://github.com/bbx10/USBH_MIDI/tree/due

Everyone with a Due is welcome to try MIDI using the native USB port. An OTG to USB host cable on the native USB port is required. Also be sure to connect external power via the barrel connector. The only testing so far is with a Korg nanoKontrol2.

YuuichiAkagawa, 2015-12-21:
@bbx10 Nice work! 
Thank you.

closed 2015-12-21 by YuuichiAkagawa

DisasterAreaDesigns, 2015-12-21:
@bbx10 Does this work on the ATSAMD / Cortex M0 / Arduino Zero processors as well?

bbx10, 2015-12-21:
@DisasterAreaDesigns No, the SAMD USB host API is different from Due and shield. I got a version to compile but no data transfers so far. The keyboard and mouse examples work so there is hope. I tried the Zero USB hub driver but it does not appear to work.

bbx10, 2015-12-29:
https://github.com/bbx10/USBH_MIDI/tree/zero

This version works on Zero and Due but there are caveats on Zero. The Zero USB host library needs more attention to fix these problems. 

WARNING ZERO 1: Debug must be on for this to work.

WARNING ZERO 2: USB host interface can block up to 10 seconds waiting for input.

WARNING ZERO 3: Do not update the USB host library. This will install the USB host library for Due which does not work on Zero. Use the Zero built-in USB host library version 1.0.0.

The Zero branch works for Due as well. The USB host library for Due does not have problems 1, 2, and 3.