YuuichiAkagawa/USBH_MIDI #27: Merge native port fork with USB host shield repo?

bbx10, 2016-01-19:
@YuuichiAkagawa 
@lazarus suggested I merge my fork with Due and Zero USB host support into https://github.com/felis/USB_Host_Shield_2.0. Does this make sense to you? My fork is your code with some changes so what do you think? I think this is a good idea though I am not sure how the build process will work.

YuuichiAkagawa, 2016-01-19:
@bbx10 @lazarus
Really?
I understand that UHS2 is not support for SAM's native.
https://github.com/felis/USB_Host_Shield_2.0/issues/193

Should I merge my code to UHS2?
I don't mind that.

bbx10, 2016-01-19:
@YuuichiAkagawa @lazarus

Please see the comments at the bottom. Lazarus wishes to start supporting Due USBHost. 

https://github.com/bbx10/USBH_MIDI/commit/bfe4c43f5ef15ced115d73440e07fdd0d674ccab

bbx10, 2016-01-19:
@YuuichiAkagawa @lauszus I summarized my results in making USBH_MIDI work on native USB host ports.

http://x10linux.blogspot.com/2016/01/arduino-usb-host-status.html

Sorry, I misspelled https://github.com/Lauszus name in my previous messages.

Lauszus, 2016-01-19:
The easiest would properly be if @YuuichiAkagawa updated his fork and sent a PR to the main repo. Then once that is merged we can merge in @bbx10's changes.

I just bought a Arduino Due and it should be on its way, then I will maintain the code in the future.

PS: @bbx10 no worries. It's not the first time ;)

Lauszus, 2016-01-19:
It would be really nice if we could merge the USB Host libraries for the Due (https://github.com/arduino-libraries/USBHost) and Zero (https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/USBHost) into 2.0, as these are just ports of  old library: https://github.com/felis/USB_Host_Shield anyway.

This should properly be possible with either just a bunch of ifdefs in Usb.cpp or we can put some specific files into separate architecture directories as it is done for the Servo library: https://github.com/arduino/Arduino/tree/master/libraries/Servo/src, but I am not really a big fan of that, as it will make it harder to maintain.

However dropping Arduino 1.0 support should not be a problem as it is really outdated by now. If you are not familiar with the Arduino 1.5 library structure, then please see: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification.

@bbx10 I will have time working on this, this coming weekend :)

Lauszus, 2016-01-19:
It would be really nice if we could merge the USB Host libraries for the Due (https://github.com/arduino-libraries/USBHost) and Zero (https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/USBHost) into 2.0, as these are just ports of  old library: https://github.com/felis/USB_Host_Shield anyway.

This should properly be possible with either just a bunch of ifdefs in Usb.cpp or we can put some specific files into separate architecture directories as it is done for the Servo library: https://github.com/arduino/Arduino/tree/master/libraries/Servo/src, but I am not really a big fan of that, as it will make it harder to maintain.

However dropping Arduino 1.0 support should not be a problem as it is really outdated by now. If you are not familiar with the Arduino 1.5 library structure, then please see: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification.

@bbx10 I will have time working on this, this coming weekend, but I don't have the hardware so I will rely on you to actually test it :)

Lauszus, 2016-01-19:
Btw @YuuichiAkagawa I can do the PR as well, but it would be better if you did it, so you will get proper credits in the Git log. I will be adding you to the following section of the readme as well: https://github.com/felis/USB_Host_Shield_2.0#developed-by.

bbx10, 2016-01-19:
@Lauszus @YuuichiAkagawa OK, I can definitely do testing for Due and Zero boards.  My SparkFun boards are on loan but they are nearly identical to the Zero so I can try them later. Merging the USBHost libraries is a complex task so I am glad a USB expert is doing it.

YuuichiAkagawa, 2016-01-19:
@Lauszus @bbx10 OK,  I  will create PR to UHS2 repo.
But I don't have DUE. I can't test.
So I want to PR my code.

Lauszus, 2016-01-19:
@bbx10 this PR should add support for the Zero via the shield: https://github.com/felis/USB_Host_Shield_2.0/pull/202. Note that I went ahead and did this, as USB and USBDevice are defined in the SAMD core, so they would have to be renamed anyway. Next step would be to add native support on top of that branch.

@YuuichiAkagawa don't worry about native support. A PR with your driver would be sufficient.

bbx10, 2016-01-19:
I forgot UHS2 + host shield hardware is a valid option for Due and Zero. I recently purchased a USB host shield so I can try these combination.

This is my understanding of the plan so far. Please correct me if I am wrong.

1. @YuuichiAkagawa submits PR to merge the existing USBH_MIDI driver to UHS2. No native USB support in either USBH_MIDI or UHS2 yet. The UHS2 commit log will show YA wrote this driver.

2. @bbx10 clones UHS2 and merges in native USB support for USBH_MIDI driver then submits a PR. However, I cannot test on Due and Zero because UHS2 does not have native USB code yet. The UHS2 commit log will show bbx10 added native support to the USBH_MIDI driver.

3. @Lauszus adds native USB support to UHS2. bbx10 tests USBH_MIDI with native USB on Due and Zero.

Lauszus, 2016-01-20:
@bbx10 yes that is correct. It would properly be easiest if you submitted your PR after I have done the one for native support.

bbx10, 2016-01-20:
@Lauszus Ok, reverse steps 2 and 3.

Lauszus, 2016-01-20:
@bbx10 exactly :)

YuuichiAkagawa, 2016-01-22:
@Lauszus
I am ready to PR.
Which branch should I send PR?

BTW current file name is 'usbh_midi.*' but class name is 'USBH_MIDI'.
Do I need to rename?

Lauszus, 2016-01-26:
@YuuichiAkagawa sorry I must have missed the email notification. Please send the PR to the master branch. The file name should be fine. You can change it if you like :+1:

YuuichiAkagawa, 2016-03-21:
Sorry, @Lauszus @bbx10 I was suspend. 
I send PR just now.

Lauszus, 2016-03-21:
@YuuichiAkagawa thanks. I just merged your PR. Can you send a PR updating the readme as well?

closed 2016-05-02 by YuuichiAkagawa