YuuichiAkagawa/USBH_MIDI #29: Why don't we get running status?

tomofu74, 2016-05-05:
Is the library have any limits?
I want to get running status, Timing Clock, from rhythm machine or sequencer.
The rhythm machine have a usb-b port and it sends realtime message and notation, velocity and so on to usb-pc host. I want to listen Timing Clock and relay to my effector's tempo pedal.
If you have any knowledge to solve my aim, please post some information.
Thanks.

YuuichiAkagawa, 2016-05-06:
I don't have a device that can send a Timing Clock.
I am testing following connection. Left side PC sent Timing Clock, Right side PC receive Timing Clock.
This test is OK.
[image]

Please test with USBH_MIDI_dump example.

tomofu74, 2016-05-07:
Thank you for your early reply.
I had done test with USBH_MIDI_dump example.  But, I can not get any response from my Arduino. ( serial monitor 115200bps Arduino IDE 1.6.8 on OS X) Nothing displayed, RX LED on Arduino UNO board is not lighten.

However, I can get any midi message in DAW software on USB. (I test in iMac OS X and iPad.)
[image]

Uhmm... My USB HOST SHIELD 2 has something wrong...?

YuuichiAkagawa, 2016-05-07:
日本語でいいですか？
このコメントを外すとデバッグメッセージが多めに出力されます。
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/usbh_midi.h#L29

その状態で、USBH_MIDI_dumpを実行したときのログを貼り付けて下さい。

Please uncomment following line. (Debug mode enable)
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/usbh_midi.h#L29

Then running again USBH_MIDI_dump and show me log.

tomofu74, 2016-05-07:
アドバイスありがとうございます。
早速、デバッグモードで走らせてみました。シリアルモニタの表示は変化なしです。残念ながらおみせできるログとして何が適切なのかわかりません。他に何か情報がお役に立てるようであれば遠慮なくお申し付けください。

I've run USBH_MIDI_dump in Debug mode.
But it is not fine. Serial monitor is unchanged. Sorry, what should I show that you need?

YuuichiAkagawa, 2016-05-07:
何も出力されないというのは少々厳しいのですが、何点か確認です。
- USB Host Shield Library 2.0のExampleにあるUSB_descでは何か表示されますか？また、マウス等別の機器を繋げた場合は何か表示されますか？
- USBHostシールドはどれを使っていますか？
- MIDI機器はバスパワーですか？自前で電源持っていますか？

tomofu74, 2016-05-07:
ご連絡ありがとうございます。
ご質問の件、順に回答いたします。

・USB_DESC を実行。起動後シリアルモニターに「Start」と表示されます。そのあと当該機器を何度か抜き差ししましたが、何も表示されませんでした。そのままさらにそのあとにiMacのキーボード、Xbox360のコントローラを差した場合は、それぞれ、Device descriptorや、Interface descriptor等、表示されました。

・USBHostシールドは、Rev2.0です。こちらで購入しました。
http://www.amazon.co.jp/USBホストシールド-Arduino-compatible-Google-Android/dp/B007YNG6RW?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00#productDetails

・MIDI機器は、自前で電源持ってます。機器は、novation社のCircuitという機器です。私の手持ちの機器では、この機器だけUSBホストシールドへの接続が認識されない、ということなのでしょうか・・・。そうすると、この機器はUSB経由でどういう通信をしているのでしょう？気になりますね。

"circuit"(novation: https://us.novationmusic.com/circuit/circuit) is NOT recognized by USB-Host-shield rev.2.0. with USB_desc example sketch. What's the matter with this product?

YuuichiAkagawa, 2016-05-08:
novationの機器は扱いが難しいイメージがあります。私もLaunchpad miniを持っていますが、USBハブを経由しないと動作しませんでした。

USBハブ経由で接続した上で、USBH_MIDI_dumpの
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USBH_MIDI_dump/USBH_MIDI_dump.ino#L36
のコメントを外して実行するとどうなるかを試して頂けないでしょうか。

tomofu74, 2016-05-08:
ご連絡ありがとうございます。
USB_descで認識されたHUBを使ってご指示のとおり試してみましたけど、残念ながらやはり何もSerial monitorには出力されませんでした。launchpad mini はUSBバスパワー動作なのでハブから給電しないと動作しないんでしょうけど、Circuitは１２Vアダプタでの動作ですのでたぶんUSB電力の問題ではないのでしょうね。

novation CIRCUIT は、USB HOST SHIELDのUSB_descですら何も表示できない機器ということから諦めざるをえないのでしょうね。残念ながらArduinoの使い道がなくなってしまいました・・・。

It's not able to be along with my hope... unfortunate.

YuuichiAkagawa, 2016-05-08:
ダメでしたか...。
※Launchpad miniは給電なしのUSBハブ経由で使えるようになる場合があります。

残念ですがUSB Host Shied Libraryで認識できない機器は、当方としてもどうにもならないので対応不可とさせてください。

npyskm, 2020-01-17:
このIssueを立てた方とは別人ですが、質問よろしいでしょうか。またここではない適切な質問場所があるなどの場合はご指摘ください。

当方、ハードウェアソフトウェア共に初心者です。
こちらの「USBH_MIDI」をArduino IDEに導入した際にサンプルコードに含まれる「USBH_MIDI_converter」を、手持ちだった「Arduino Pro Mini 328 3.3V 8MHz ＋ ミニUSBホストシールド (＋ MIDIアウト回路)」上で動かすことで、「KORG nanoKEY Studio」をUSBキーボードとして使えないかと考えています。

[質問]
nanoKEY無印は対応機種に含まれていますが、Studioは含まれていないことは理解しています。そのそもそもの意味としては、ハードウェアの何らかのIDの制限などの理由で、対応機種に含まれていないものは全く動かないという話なのでしょうか、それともソフトウェアを多少触ることによって動くようになる感じでしょうか、またはおおかたは動くはずなのでしょうか。

[試したこと]
1)「USBH_MIDI_converter」に、LED点灯を使った簡易デバッグをすることで、Usb.Init()は通過することはわかった(ミニUSBホストシールドでも使える、と理解)が、Midi.RecvData(outBuf)は出ていない様子。
2)nanoKEY StudioからMIDIとして信号を受け取れていないのかと思い「USBH_MIDI_dump」を書き込んだが、「シリアルモニタ」窓を開いても何も表示されず(使い方が合っているか自信無し)。

YuuichiAkagawa, 2020-01-17:
日本語で構わないので、新しいissueでお願いします。

npyskm, 2020-01-17:
失礼しました、承知しました。

YuuichiAkagawa, 2020-11-23:
It's a long time ago, but this issue has been resolved.   
https://github.com/felis/USB_Host_Shield_2.0/issues/563

closed 2020-11-23 by YuuichiAkagawa