YuuichiAkagawa/USBH_MIDI #30: midi filter

lokkiikkol, 2016-05-13:
hi, i am trying to remap the keys of an axis-49 controller. the connection via usb-shield works flawlessly, perfect! the controller runs in selfless mode, which means it outputs a note on (and off) from 1 to 98 for every key respectively. i want to remap those note numbers to different ones. so nr. 1 will be 18, 2 20, 3, 22 and so on. i created an array remap[ ] that stores all the values. looking at the simple example (usb midi converter) i thought i could do something like: outBuf[2] = remap[outBuf[2]]; to remap all the keys, but it does not work. is there a way to do this with this library? cheers

YuuichiAkagawa, 2016-05-13:
I was test a note remap. It's work.
```C++
uint8_t remap[] = {
   0, 18, 20, 22, 24, 26, 28, 30,
  19, 21, 23, 25, 27, 29, 31, 33,
   2,  4,  6,  8, 10, 12, 14, 16,
   1,  3,  5,  7,  9, 11, 13, 15
};
// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
  uint8_t outBuf[ 3 ];
  uint8_t size;

  do {
    if ( (size = Midi.RecvData(outBuf)) > 0 ) {
      //MIDI Output
      outBuf[1] = remap[outBuf[1] - 0x30];  //nanoKEY's note# is 0x30-48
      _MIDI_SERIAL_PORT.write(outBuf, size);
    }
  } while (size > 0);
}
```

Connection is like this.
 nanoKEY ---USB---> Arduino ---MIDI---> Roland QuadCapture ---USB---> PC (MIDI-OX)

lokkiikkol, 2016-05-13:
thanks for the fast reply! that is exactly what i did, with one small difference. i put the remapping before the if statement, which makes of course a difference. i was tired :) i will test as soon as i get home and will report back.

lokkiikkol, 2016-05-13:
and it works perfectly now! great, thanks so much!!!

closed 2016-05-13 by lokkiikkol

YuuichiAkagawa, 2016-05-14:
:smile: