YuuichiAkagawa/USBH_MIDI #32: Launchpad mini

fortunatop, 2016-07-22:
Hello, i have troubles trying to get a Novation Launchpad Mini running. It's supposed to be midi compliant, but is not working if i declare it as a midi usb device.
if i run the hardware test without declaring it as a midi device (using just the usb hub 2.0 library) it runs with no issues (gets to the running status), but if i use the usbh_midi and declare Midi(Usb) it gets stuck in waiting for first SOF status. Any clues on how to solve it? 
The test runs ok with a waldorf rocket synth. Actually, been sequencing it with the arduino via usb host with no issues with your code, great work btw. 
Thanks in advance.

YuuichiAkagawa, 2016-07-26:
novation's product is delicate.
Please check following issue.
https://github.com/YuuichiAkagawa/USBH_MIDI/issues/21

- Do you have enough power ?
- Do you try to connect via USB HUB?

fortunatop, 2016-07-26:
Thanks for your reply.
i've been doing some more tests and actually the problem is with usb host 2.0 i think.
i run the descriptor example in the library. With the rocket synth several list of variables are shown, while when trying with the launchpad, only the first genreal descriptor i shown and then something gets bugged. guess i'll trade the launchpad for an akai apc mini. Thanks again for your concern.

closed 2016-11-13 by fortunatop