YuuichiAkagawa/USBH_MIDI #34: Arturia beatstep pro

fortunatop, 2016-11-13:
hello, been trying to use an arduino with usb host cor converting the usb midi out of an Arturia beatstep pro to midi din, using the USB midi converter with no luck. eems like no midi streaming is geting to the beard. here's the USB desc

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1C75
Product ID:		0287
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0085
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		24
Contents:	010001090001010904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	0100015F000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020101040624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020202000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020105060624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020206000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030103010201030924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030204010101000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030107010601050924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030208010501000905

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	00
Unknown descriptor:
Length:		06
Type:		25
Contents:	010201050905

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	00
Unknown descriptor:
Length:		06
Type:		25
Contents:	01020307140D

Addr:1(0.0.1)

Runnig the USB midi dump i get this:

VID:1C75, PID:0287

fortunatop, 2016-11-13:
never mind, is working great. declaring a usb hub solved it. 
guess you can add the beatstep pro to working devices.

Thanks a lot for this.

closed 2016-11-13 by fortunatop

YuuichiAkagawa, 2016-11-13:
Thank you for your report. 😄