YuuichiAkagawa/USBH_MIDI #35: Question: what's the proper way to handle incoming sysex

neroroxxx, 2017-01-10:
Hi, first of all thank you for awesome library!

I have a project that sends and receives sysex, while sending with your library is flawless, sometimes when i receive messages (specially when it's 2 or more) they don't have the 0xF0, the rest of the bytes are there but no 0xF0, i'm using this with a guitar processor which when i monitor it on my mac it's sending everything properly so i know it's something i'm not doing right.

I wrote some code based on your examples but again for some reason sometimes i don't get the 0xF0 and in those ocassions sometimes there are bytes missing or a byte added here and there.

Do you have any other examples to receive a sysex array properly?

Thanks!

YuuichiAkagawa, 2017-01-11:
How long your sysex message length?

neroroxxx, 2017-01-14:
Hi! my sysex go from 12bytes to 100 bytes sometimes

Thanks!

YuuichiAkagawa, 2017-01-20:
Sorry for the delay in response.
Current implementation buffers until the end of SysEx. And buffer size is only 64 bytes.

I added a new function experimentally. It can process every packet.
Note: If another packet enters in the SysEx stream, it will be broken.

Please test feature/sysex#2  branch.
https://github.com/YuuichiAkagawa/USBH_MIDI/tree/feature/sysex%232

Example
https://gist.github.com/YuuichiAkagawa/959002dea3e7ad94c09c76e6fdafb8bc

Or if you have enouth memory, you can increase buffer size.
- MIDI_SYSEX_BUFFER_SIZE
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/feature/sysex%232/usbh_midi.h#L36

neroroxxx, 2017-01-20:
Thanks! I’ll give it a shot! is this compatible with USB Host 2.0 or is it going to be for 3.0?

Thank you again!

YuuichiAkagawa, 2017-01-20:
It is compatible with UHS2. UHS3 support is not yet.

YuuichiAkagawa, 2017-02-22:
SysEx code was updated. Please check v0.3.2.

neroroxxx, 2017-02-24:
It works much much better! i'll keep testing and see if it continues to work as expected. Thanks!

YuuichiAkagawa, 2017-02-24:
😃

closed 2017-02-24 by YuuichiAkagawa