YuuichiAkagawa/USBH_MIDI #38: Cannot get Launchpad S to run

jkruseAUT, 2017-10-29:
I have tried a number of things.

Here is the Debug output:

MIDI Init
Addr:01
VID:1235 PID:0020 #Conf:01

Conf:01 Int:00 Alt:00 EPs:00 IntCl:01 IntSubCl:01
No MIDI Device

Conf:01 Int:01 Alt:00 EPs:02 IntCl:01 IntSubCl:03
MIDI Device
-EPAddr:81 bmAttr:03 MaxPktSz:40
-EPAddr:02 bmAttr:03 MaxPktSz:40

NumEP:01

But the Launchpad S does not work. I looked at Usb.getUsbTaskState()  and the result is 160 (DEC) which is USB_STATE_ERROR (0xA0 in HEX).

YuuichiAkagawa, 2017-10-29:
novation's product is delicate.
Please check following issue.
#21 
- Do you have enough power ?
- Do you try to connect via USB HUB?

jkruseAUT, 2017-10-29:
I tried the following today:
- use 12V on Arduino input
- use USB 2.0 Hub (powered)

Still no luck, but I get the following output from USB_MIDI_DUMP now:

MIDI Init
Addr:0B
VID:1235 PID:0020 #Conf:01

Conf:01 Int:00 Alt:00 EPs:00 IntCl:01 IntSubCl:01
No MIDI Device

Conf:01 Int:01 Alt:00 EPs:02 IntCl:01 IntSubCl:03
MIDI Device
-EPAddr:81 bmAttr:03 MaxPktSz:40
-EPAddr:02 bmAttr:03 MaxPktSz:40

NumEP:01
VID:1235, PID:0020

MIDI Init
Addr:0C
VID:05E3 PID:0608 #Conf:01

Conf:01 Int:00 Alt:00 EPs:01 IntCl:09 IntSubCl:00
No MIDI Device
-EPAddr:81 bmAttr:03 MaxPktSz:01

NumEP:01
VID:05E3, PID:0608

But when I press any key/pad nothing happens. It does not seem to read midi or it does not listen on the right channel maybe?

Any idea, much appreciated :)

jkruseAUT, 2017-10-29:
Maybe I am using the wrong address?   Or cable number?  What does that mean? How do I send a simple Midi Command?

YuuichiAkagawa, 2017-10-30:
I found problem. 
https://github.com/felis/USB_Host_Shield_2.0/issues/313 is a degradation.
LaunchPad has illegal endpoint attribute.

bmAttributes.TransferType
- valid : 2(Bulk)
- launchpad : 3 (Interrupt)

Please use this project's usbh_midi.cpp.

Thank you.

jkruseAUT, 2017-10-30:
It works!  Even without the hub - I am only using 12V on Arduino to power Novation Launchpad S and Novation Impulse25.

THANK YOU VERY MUCH! I really appreciate your help.

YuuichiAkagawa, 2017-10-31:
I want descriptor info.
Please run USB_desc example and show me result.

jkruseAUT, 2017-10-31:
01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1235
Product ID:		0020
Revision ID:		0000
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0052
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		09
Type:		24
Contents:	010001090001010904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	0100012E000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020101000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030102010101000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	01
Unknown descriptor:
Length:		05
Type:		25
Contents:	0101020705

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0040
Polling interval:	01
Unknown descriptor:
Length:		05
Type:		25
Contents:	010101AFF7

Addr:1(0.0.1)

YuuichiAkagawa, 2017-10-31:
Thanks for your cooperation.

commit b02329b5 (2018-03-24): Added Vendor/Device specific setup support (fixed #38)
commit 2b82ab64 (2018-03-24): Added Vendor/Device specific setup support (fixed #38)
closed 2018-03-24 by YuuichiAkagawa