YuuichiAkagawa/USBH_MIDI #39: Connecting Arduino Micro 32u4 USB/MIDI keyboard to USB Host Shield

knowsbetter, 2018-07-03:
Hello!
I'm trying to connect Arduino Micro 32u4 USB/MIDI keyboard to USB Host Shield. Micro is programmed using MIDIUSB library. I want to recieve MIDI messages from the keyboard and handle them with Arduino UNO the Shield is mounted on. But i can't get it to work. It seems like the Shield doesn't recieve anything and Micro's LED doesn't even blink as it always does during MIDI messages sending. What should I start with to solve the problem?

YuuichiAkagawa, 2018-07-04:
Is your USB Host Shield works normally ? Please try USB_desc.

knowsbetter, 2018-07-04:
Yes, it works well with a mouse and ADK.
USB_desc output:

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		2341
Product ID:		8037
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		00AF
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA
Unknown descriptor:
Length:		08
Type:		0B
Contents:	0002020201000904

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		01
Intf. Class:		02
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		05
Type:		24
Contents:	0010010524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0101010424
Unknown descriptor:
Length:		04
Type:		24
Contents:	02060524
Unknown descriptor:
Length:		05
Type:		24
Contents:	0600010705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	40

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		0A
Intf. Subclass:		00
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00
Unknown descriptor:
Length:		08
Type:		0B
Contents:	0202010100000904

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		24
Contents:	010001090001030904

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01000141000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020101000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020202000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030103010201000924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030204010101000905

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0040
Polling interval:	00
Unknown descriptor:
Length:		05
Type:		25
Contents:	0101010905

Endpoint descriptor:
Endpoint address:	85
Attr.:			02
Max.pkt size:		0040
Polling interval:	00
Unknown descriptor:
Length:		05
Type:		25
Contents:	01010336FE

It has a few additional endpoints comparing to another Arduino Micro programmed with a simple Blink sketch.

YuuichiAkagawa, 2018-07-05:
Thank you for your report.
I don't have Arduino Micro. I tested using Leonardo.

Leonardo:
-  MIDIUSB_write (MIDIUSB 1.0.3)

Uno+UHS2:
- USBH_MIDI_dump (USBH_MIDI 0.4.0)

It seems work fine.
[image]
Uno can receive note on/off messages.
[image]

knowsbetter, 2018-07-05:
Thank you for your response! I'll try it using Leonardo and let you know. There might be something wrong with my Host shield. Because Leonardo and Micro both are 32u4.

knowsbetter, 2018-07-20:
Dear YuuichiAkagawa,
thank you for your support! Everything's fine now. There was something wrong with a copy of USBH library on my PC. I might have edited a few files and forgot about it. Solved!

closed 2018-07-20 by knowsbetter