YuuichiAkagawa/USBH_MIDI #41: Novation Launchkey Mini Issue

FUCKYOURBRAIN, 2019-02-01:
Hello,

Usb.getUsbTaskState() returns 0xa0 (USB_STATE_ERROR), when Novation Launchkey Mini controller is connected. 

USB_desc returns the following:

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1235
Product ID:		0035
Revision ID:		0000
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0063
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		09
Type:		24
Contents:	010001090001010904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	0100013F000624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020101040624
Unknown descriptor:
Length:		06
Type:		24
Contents:	020104050924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030102010101040924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030103010202050705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	01
Unknown descriptor:
Length:		06
Type:		25
Contents:	010202030705

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0040
Polling interval:	01
Unknown descriptor:
Length:		06
Type:		25
Contents:	01020104000B

Addr:1(0.0.1)

Maybe it's worth mentioning, that when I connect the controller, it shows a starting pads backlight sequence and after that pads backlight respond on touch. So I assume, that problem is not in shield library or the controller itself.

I checked all novation issues and tried the following:

- Connecting arduino to 12V (nothing changes)
- Inserting a powered USB 2.0 HUB in between arduino and midi controller (staring backling sequence loops over and over, but still no midi data)

Thanks in advance,
Hope for your help!

YuuichiAkagawa, 2019-02-02:
It is side effect of this fix.
https://github.com/felis/USB_Host_Shield_2.0/issues/313

Please change the following line.
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/usbh_midi.cpp#L350

```
                if(pid == 0x20 || pid == 0x35 || pid == 0x36 || pid == 0x51 || pid == 0x69 ) {
```

FUCKYOURBRAIN, 2019-02-02:
Thanks a lot! I works!

Today I've stumbled with a fact, that I can't send MIDI data to the controller.
I undrestand, that the queston is not connected with the current issue directly, but mabe you can help.

I'm trying to turn on pad backlight in accordance with the controller manual:

> To light the bottom left drum-pad the following message would be sent
> Example LED on message: MIDI Channel 16, C1, Colour/Velocity (1-127)
> 9Fh, 24h, Colour/Velocity (01h - 7Fh)
> (159, 36, Colour/Velocity (1- 127))

So I'm just checking all possible velocities for a single pad, but no luck:

```
void loop()
{
  uint8_t msg[3] = {159, 36, 0};
  Usb.Task();
  uint32_t t1 = (uint32_t)micros();
  if ( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {    
    for (int i=0; i<128; i++)
    {        
       Serial.println(i);
       msg[2] = i;
       Midi.SendData(msg, 3);
       delay(100);
    }
  }
}

```
Identical code on python, when the controller is connected to PC, changes backlight successfully.

Please, tell me what I'm doing wrong.

closed 2019-02-02 by FUCKYOURBRAIN