YuuichiAkagawa/USBH_MIDI #42: Can't send MIDI data to the controller

FUCKYOURBRAIN, 2019-02-09:
II've stumbled with a fact, that I can't send MIDI data to the controller.
I'm trying to turn on pad backlight in accordance with the controller manual:

> To light the bottom left drum-pad the following message would be sent
> Example LED on message: MIDI Channel 16, C1, Colour/Velocity (1-127)
> 9Fh, 24h, Colour/Velocity (01h - 7Fh)
> (159, 36, Colour/Velocity (1- 127))

So I'm just checking all possible velocities for a single pad, but the controller doesn't respond:

```
void loop()
{
  uint8_t msg[3] = {159, 36, 0};
  Usb.Task();
  uint32_t t1 = (uint32_t)micros();
  if ( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {    
    for (int i=0; i<128; i++)
    {        
       Serial.println(i);
       msg[2] = i;
       Midi.SendData(msg, 3);
       delay(100);
    }
  }
}
```
Identical code on python, when the controller is connected to PC, changes backlight successfully.

Please, tell me what I'm doing wrong.

YuuichiAkagawa, 2019-02-10:
You need to execute a Usb.Task() for each Midi.SendData().
```
void loop()
{
  uint8_t msg[3] = {159, 36, 0};
  Usb.Task();
  if( Midi ) {
    for (int i=0; i<128; i++)
    {        
       Serial.println(i);
       msg[2] = i;
       Midi.SendData(msg, 3);
       Usb.Task();
       delay(100);
    }
  }
}
```

FUCKYOURBRAIN, 2019-02-10:
Unfortunatelly, controller doesn't respond.

I've tried adding external power to the Arduino, or using a powered USB hub, but it's all the same.
I found a project with USB Host Shield + Novation Launchpad (a similar conroller) on web, and took SendData function call from that project.

But  аm I using SendData function correctly?

```
uint8_t msg[3] = {159, 12, 127};
Midi.SendData(msg, 3);
Usb.Task();
```

I can't get what is a nCable parameter for:

```
/* Send data to MIDI device */
uint8_t USBH_MIDI::SendData(uint8_t *dataptr, uint8_t nCable)
```

YuuichiAkagawa, 2019-02-10:
I was test with Lanchpad mini.  And I use with USB-HUB too.
https://youtu.be/KUsNGOG79JM
```
#include <usbh_midi.h>
#include <usbhub.h>
#include <SPI.h>

USB Usb;
USBHub Hub(&Usb);
USBH_MIDI  Midi(&Usb);

void setup()
{
  Serial.begin(115200);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

uint8_t i = 0;
void loop()
{
//  uint8_t msg[3] = {144, 36, 0};  //LaunchPad channel 1
  uint8_t msg[3] = {159, 36, 0};  //LaunchKey channel 16
  Usb.Task();
  if( Midi ) {
       //Receive from MIDI controller
       MIDI_poll();

       //Send to MIDI controller
       msg[2] = i;
       Midi.SendData(msg, 3);
       i++;
       delay(100);
  }
}

// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
  uint8_t outBuf[ 3 ];
  uint8_t size;

  do {
    if ( (size = Midi.RecvData(outBuf)) > 0 ) {
      //MIDI Output
      Serial.print("RecvData: ");
      Serial.println(outBuf[1]);
    }
  } while (size > 0);
}
```
I don't know how to work LaunchKey.

The meaning of nCable is MIDI port number. It is not a channel.

FUCKYOURBRAIN, 2019-02-10:
> The meaning of nCable is MIDI port number. It is not a channel.

This helped A LOT. Maybe because of "InControl" feature, on Launchkey the port number is different:

```
void loop()
{
  uint8_t inControlOn[3] = {144, 12, 127};
  uint8_t inControlOff[3] = {144, 12, 0};
  uint8_t pad1_On[3] = {144, 96, 15};
  uint8_t pad2_On[3] = {144, 97, 19};
  uint8_t pad3_On[3] = {144, 98, 21};
  uint8_t pad4_On[3] = {144, 99, 24};
  uint8_t rc;
  Usb.Task();
  if( Midi ) {
       /* inControl mode on */
       rc = Midi.SendData(inControlOn, 1); 
       delay(100); 

       rc = Midi.SendData(pad1_On, 1); 
       delay(100); 
       rc = Midi.SendData(pad2_On, 1); 
       delay(100); 
       rc = Midi.SendData(pad3_On, 1); 
       delay(100); 
       rc = Midi.SendData(pad4_On, 1); 
       delay(100); 
       
       /* inControl mode off */
       rc = Midi.SendData(inControlOff, 1); 
       delay(100);
  }
}
```

Thanks for your time!

closed 2019-02-10 by FUCKYOURBRAIN