YuuichiAkagawa/USBH_MIDI #43: Note-off messages are being dropped.

cdw2000, 2019-02-26:
Based on Issue #9, this appears to be an old problem that was fixed, but I am having this problem using my M-Audio Oxygen 49 keyboard.   Single notes are fine, but playing multiple notes simultaneously in a repetitive manner sometimes causes a note to be stuck on.

I'm using an Arduino Uno R3 with a USB Host Shield 2.0 along with the current Arduino IDE, MIDI library 4.3.1, USB Host Shield library 2.0, and your current USBH_MIDI release.   I'm using your USB_MIDI_converter example sketch.   Based on reading Issue #9 should I be using the USB_MIDI_converter_multi sketch instead?

YuuichiAkagawa, 2019-02-28:
Please run "USBH_MIDI_dump" example.

cdw2000, 2019-02-28:
I will do that tonight (EST) and post the results.

cdw2000, 2019-03-02:
Here is my dump file.   I basically played a C chord repeatedly.  Since the USB MIDI Converter code uses the TXO on the Arduino for the MIDI out, I couldn't connect my MIDI sound device (Sparkfun WAV Trigger) board to hear if any of the notes got stuck on.   Hopefully I captured what you need.
[Oxygen 49 Dump.txt](https://github.com/YuuichiAkagawa/USBH_MIDI/files/2921490/Oxygen.49.Dump.txt)

YuuichiAkagawa, 2019-03-02:
Please try to remove or reduce the delay time.
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USB_MIDI_converter/USB_MIDI_converter.ino#L76

```
  //delay(1ms)
  //doDelay(t1, (uint32_t)micros(), 1000);
  //or
  doDelay(t1, (uint32_t)micros(), 100);
```

cdw2000, 2019-03-02:
OK. I'll give that a try and will let you know the results.

cdw2000, 2019-03-02:
Changing the timing worked.  Thanks for your help.   I used 480 usec.   Here's my rational:   MIDI data rate is 31250 bits/sec.  A 3 byte MIDI message (Note On/Off, Note No., Velocity) is 30 bits.   31250/30 is 1041.67 messages/second.   Taking the reciprocal yields one message possible every 960 usec.   Setting  the sample interval at half that (480 usec) would ensure no messages lost.

closed 2019-03-03 by YuuichiAkagawa