YuuichiAkagawa/USBH_MIDI #44: Yamaha THR10, Midi with VendorClass (255) , Buffer overflow

martinzw, 2019-03-29:
Hi ,
finally  i got my "Yamaha THR10" (guitar amp, that can be configured via MIDI-SysEx) working with this library and my "Arduino Due", but there were several problems to solve before:
- USB-Enumeration works only, if Debug-outputs are enabled in USBHost.h

```
  #define TRACE_USBHOST(x)	x
//#define TRACE_USBHOST(x)
```

 otherwise device is not detected. Don't ask me why... (perhaps a timing problem, that is solved when a "printf" delays process at the right time?)
- Adding an "else if"  in descriptor parsing
```      case  USB_DESCRIPTOR_INTERFACE :
        if( buf_ptr[5] == USB_CLASS_AUDIO && buf_ptr[6] == USB_SUBCLASS_MIDISTREAMING )
	{  //p[5]; bInterfaceClass = 1(Audio), p[6]; bInterfaceSubClass = 3(MIDI Streaming)
          isMidiFound = true; //MIDI device found.
          isMidi      = true;
        }
//**Yamaha THR10**
	else if( buf_ptr[5] == USB_CLASS_VENDOR_SPECIFIC && buf_ptr[6] == USB_SUBCLASS_MIDISTREAMING ) //**Yamaha THR10**
		{  
                   //p[5]; bInterfaceClass = 255(VENDOR), p[6]; bInterfaceSubClass = 3(MIDI Streaming)
		   isMidiFound = true; //MIDI device found.
		   isMidi      = true;
		}
		else
		{
#ifdef DEBUG
          Serial.print("No MIDI Device\n");
#endif
//          buf_ptr += total_length + 1;
//          bConfNum = 0;
            isMidi = false;
        }
        break;
```

- Solving buffer problems:
1) always use` (uint16_t)` cast instead of (uint8_t) cast for `epInfo[index].maxPktSize` 
e.g. 
```
  // Extract Max Packet Size from device descriptor
  epInfo[0].maxPktSize = (uint16_t)((USB_DEVICE_DESCRIPTOR*)buf)->bMaxPacketSize0; 
```
             
2) quick and dirty change for buffersize, that works for THR10 (just a work around, no proper solution, i guess!!!):
```
if( isMidi )
{
            if ((epDesc->bEndpointAddress & 0x80) == 0x80) 
   	   {
              // Input
              index = epDataInIndex;
              epInfo[index].epAddr	= (epDesc->bEndpointAddress & 0x0F);
              //epInfo[index].maxPktSize	= (uint16_t)epDesc->wMaxPacketSize;
	      epInfo[index].maxPktSize	= (uint16_t)2048;  //necessary for YAMAHA THR10
              pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].epAddr, UOTGHS_HSTPIPCFG_PTYPE_BLK, UOTGHS_HSTPIPCFG_PTOKEN_IN, epInfo[index].maxPktSize, 0, UOTGHS_HSTPIPCFG_PBK_1_BANK);
            }
            else 
	    {
              // Output
              index = epDataOutIndex;
              epInfo[index].epAddr	= (epDesc->bEndpointAddress & 0x0F);
              //epInfo[index].maxPktSize	= (uint8_t)epDesc->wMaxPacketSize;
			  epInfo[index].maxPktSize	= (uint16_t)2048;  //necessary for YAMAHA THR10
              pipe = UHD_Pipe_Alloc(bAddress, epInfo[index].epAddr, UOTGHS_HSTPIPCFG_PTYPE_BLK, UOTGHS_HSTPIPCFG_PTOKEN_OUT, epInfo[index].maxPktSize, 0, UOTGHS_HSTPIPCFG_PBK_1_BANK);
            }
```

After that modifikations, i can send System-Exlusive Messages to THR10, like 

```
uint8_t msg1[]= {0xF0,0x43,0x7D,0x30,0x41,0x30,0x01,0x01,0xF7};  //Lamp off
uint8_t msg2[]= {0xF0,0x43,0x7D,0x30,0x41,0x30,0x01,0x00,0xF7};  //Lamp on`
```
that switches the light of the Guitar-Amp.

But more important, I can send sound-Patches,  that are created as a file before with the THR10-Editor on PC.

This way I can build a foot-switch / pedal to change sound patches on the fly like 

[THR10 foot switch](http://www.mehlbrandt.de/thr/thr_11.html)

But this guy did it with an ordinary Arduino Uno and the USB-Host Shield.
With the Due I can use the native USB-Host-Port, have got more memory and speed.
I want to inject Samples to the THR10 with this footswitch as well, as it can also act as IN-/OUT Audio Interface.

YuuichiAkagawa, 2019-03-30:
Thank you for your information.
This library is work with USB_Host_Shield_2.0, not ArduinoCore-samd library.

I changed the title for this issue.

martinzw, 2019-03-30:
OK, sorry, my fault...
My Issue was in fact targeted to the [fork USBH_MIDI, branch "Due"](https://github.com/bbx10/USBH_MIDI/tree/due) not to your shield library, that bbx10 forked from.
This one works with Due and Zero using their native USB-host port, not with the shield.

It is a pitty, that ther is no unified library for all boards with or without use of a USB host shield.

closed 2019-03-31 by YuuichiAkagawa