YuuichiAkagawa/USBH_MIDI #45: Sysex dump > 64 bytes

DarthDev, 2019-04-02:
Hello

Is it possible to receive a SysEx dump that is greater than 64 bytes? I can't seem to find a way to achieve this. Any assistance would be greatly appreciated.

Many thanks
DarthDev

YuuichiAkagawa, 2019-04-07:
USB MIDI Packet size is 64 bytes.
If you want to receive more than 64 bytes of data,  you need to iteration.

```
#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

#define _MIDI_SERIAL_PORT Serial

USB Usb;
USBH_MIDI  Midi(&Usb);

void setup()
{
  _MIDI_SERIAL_PORT.begin(115200);

  //Workaround for non UHS2.0 Shield
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  Usb.Task();
  uint32_t t1 = (uint32_t)micros();
  if ( Midi ) {
    MIDI_poll();
  }
}

// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll()
{
  uint8_t size;
  uint8_t outBuf[4];
  uint8_t sysexBuf[3];
  char buf[20];

  do {
    if ( (size = Midi.RecvRawData(outBuf)) > 0 ) {
      uint8_t rc = Midi.extractSysExData(outBuf, sysexBuf);
      if ( rc != 0 ) { 
        //SysEx
        _MIDI_SERIAL_PORT.print("SysEx:");
        for (int i = 0; i < rc; i++) {
          sprintf(buf, " %02X", sysexBuf[i]);
          _MIDI_SERIAL_PORT.print(buf);
        }
        
      } else {
        _MIDI_SERIAL_PORT.print("_____:");
        for (int i = 1; i <= size; i++) {
          sprintf(buf, " %02X", outBuf[i]);
          _MIDI_SERIAL_PORT.print(buf);
        }
      }
      _MIDI_SERIAL_PORT.println();
    }
  } while (size > 0);
}
```

DarthDev, 2019-04-09:
Thanks very much for your help. I'm still struggling though. I'm not getting the whole SysEx dump. The first 64 bytes are fine, but then the following three bytes are 06 01 F7, which means "end of SysEx message", if I understand correctly. The next set of bytes is some sort of signal that my MIDI device uses to let the PC know it is still connected - the remainder of the SysEx dump seems to have vanished.

YuuichiAkagawa, 2019-04-09:
Umm...
I was test with following connection.

> PC-- (USB) --> Roland Quad-Capture --(MIDI)--> Roland UM-ONE --(USB)--> UHS+Arduino

I received a 1000 byte SysEx message from PC.

jsmmns2000, 2020-06-02:
Hello,
i am using this library to send sysex commands from an Arduino MKRzero to a Zoom G1xon pedal over USB. I can send program change commands successfully.
If i send the generic ID request **F0 7E 00 06 01 F7**, i get the correct reply **F0 7E 00 06 02 52 64 00 01 00 31 2E 32 31 F7**,
but when i send **F0 52 00 64 09 00 00 02 F7** (request patch 02) , i get a message that misses random bytes.
The correct reply should be **F0 52 00 64 08 00 00 02 70 00 00 11 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 21 00 00 10 48 0B 10 01 64 00 45 22 26 48 08 02 00 00 06 00 00 21 00 00 12 03 40 08 00 2E 20 06 2C 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 60 00 64 01 00 00 00 20 03 00 00 00 00 42 6C 75 65 00 20 4C 65 61 64 20 00 07 37 39 77 0C F7** (144 bytes)

I usually receive from 90 to 132 bytes.

I 'm using the code above (for the iteration) in which i have added a button in the loop() that sends the command end then goes for Usb.Task(); and MIDI_poll(); staff. I tried with some delay after sending the command but i get the same behavior.

Am i missing something? Can you suggest anything?

Using amidi i can successfully send and receive the commands.

Thanks

YuuichiAkagawa, 2020-06-20:
Hello, @jsmmns2000
I bought G1on.
> but when i send F0 52 00 64 09 00 00 02 F7 (request patch 02) , i get a message that misses random bytes.

I sent **F0 52 00 63 09 00 00 02 F7**  and received the correct result.
I used with Arduino Uno.

## Code
```
#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

#define _MIDI_SERIAL_PORT Serial

uint8_t genericid[] {
  0xF0, 0x7E, 0x00, 0x06, 0x01, 0xF7
};

uint8_t reqpatch2[] {
  0xF0, 0x52, 0x00, 0x63, 0x09, 0x00, 0x00, 0x02, 0xF7 
};

USB Usb;
USBH_MIDI  Midi(&Usb);
uint8_t phase = 0;

void setup()
{
  _MIDI_SERIAL_PORT.begin(115200);

  //Workaround for non UHS2.0 Shield
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  Usb.Task();
  if ( Midi ) {
    MIDI_poll();
    if( phase == 0 ){
        _MIDI_SERIAL_PORT.println("\nRequest: Generic ID");
        Midi.SendSysEx(genericid, sizeof(genericid));
        phase++;
        delay(10);
    }else if( phase == 1 ){
        _MIDI_SERIAL_PORT.println("\nRequest: Patch 02");
        Midi.SendSysEx(reqpatch2, sizeof(reqpatch2));
        phase++;
        delay(10);
    }
  }
}

// Poll USB MIDI Controler
void MIDI_poll()
{
  uint8_t size;
  uint8_t outBuf[4];
  uint8_t sysexBuf[3];
  char buf[20];

  do {
    if ( (size = Midi.RecvRawData(outBuf)) > 0 ) {
      uint8_t rc = Midi.extractSysExData(outBuf, sysexBuf);
      if ( rc != 0 ) { 
        //SysEx
        _MIDI_SERIAL_PORT.print("SysEx:");
        for (int i = 0; i < rc; i++) {
          sprintf(buf, " %02X", sysexBuf[i]);
          _MIDI_SERIAL_PORT.print(buf);
        }
        
      } else {
        _MIDI_SERIAL_PORT.print("_____:");
        for (int i = 1; i <= size; i++) {
          sprintf(buf, " %02X", outBuf[i]);
          _MIDI_SERIAL_PORT.print(buf);
        }
      }
      _MIDI_SERIAL_PORT.println();
    }
  } while (size > 0);
}
```
## Result
```
Request: Generic ID
SysEx: F0 7E 00
SysEx: 06 02 52
SysEx: 63 00 00
SysEx: 00 31 2E
SysEx: 32 31 F7

Request: Patch 02
SysEx: F0 52 00
SysEx: 63 08 00
SysEx: 00 02 70
SysEx: 00 00 11
SysEx: 00 00 42
SysEx: 00 38 00
SysEx: 00 64 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 21 00
SysEx: 00 10 48
SysEx: 0B 10 01
SysEx: 64 00 45
SysEx: 22 26 48
SysEx: 08 02 00
SysEx: 00 06 00
SysEx: 00 21 00
SysEx: 00 12 03
SysEx: 40 08 00
SysEx: 2E 20 06
SysEx: 2C 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 01 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 01 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 00
SysEx: 00 00 60
SysEx: 00 64 01
SysEx: 00 00 00
SysEx: 20 03 00
SysEx: 00 00 00
SysEx: 42 6C 75
SysEx: 65 00 20
SysEx: 4C 65 61
SysEx: 64 20 00
SysEx: 59 54 05
SysEx: 40 0D F7
```

YuuichiAkagawa, 2020-07-26:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2020-07-26 by YuuichiAkagawa