YuuichiAkagawa/USBH_MIDI #46: sending midi clock to host

lokkiikkol, 2019-06-05:
hi, i try to send midi clock messages (248) to a waldorf blofeld. the messages should be retrieved on port 2 on the blofeld, what function should i use? i tried this simple test code and something is received by the blofeld, but the tempo does not sync to the Arduino messages:

#include <usbh_midi.h>
#include <usbhub.h>
#include <SPI.h>

USB Usb;
USBHub Hub(&Usb);
USBH_MIDI  Midi(&Usb);
uint8_t start = 0;
void setup()
{
  Serial.begin(115200);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  uint8_t msg2 = 250;
  uint8_t msg = 248;
  //LaunchKey channel 16
  if (!start) {
    Usb.Task();
    Midi.SendData(msg2,1);
    start = 1;
  }
  Usb.Task();

       Midi.SendData(msg,1);
       delay(10);
  }

YuuichiAkagawa, 2019-06-05:
Please try this code.
```
#include <usbh_midi.h>
#include <usbhub.h>
#include <SPI.h>

USB Usb;
USBHub Hub(&Usb);
USBH_MIDI Midi(&Usb);
uint8_t start = 0;
void setup()
{
    Serial.begin(115200);

    if (Usb.Init() == -1) {
        while (1); //halt
    }//if (Usb.Init() == -1...
    delay( 200 );
}

uint8_t msg2 = 250;
uint8_t msg1 = 248;
uint8_t msg[3];

void loop()
{
  Usb.Task();
  if( Midi ) {
    if (!start) {
        msg[0] = msg2;
        Midi.SendData(msg);
        start = 1;
        Usb.Task();
    }
    msg[0] = msg1;
    Midi.SendData(msg);
    delay(10);
  }
}
```
I am testing with MIDI-OX. It seems to work fine.

The second argument of SendData is "Cable number". It is not a size.

lokkiikkol, 2019-06-05:
i will try. i put a 2 as a second argument, because i have to send to the
second port on the blofeld, but maybe that is not the same as cable select?

lokkiikkol, 2019-06-05:
i get this when i try to compile your code:
``
Arduino: 1.8.5 (Mac OS X), Board: "Arduino/Genuino Uno"

/Applications/new arduino/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/new arduino/Arduino.app/Contents/Java/hardware -hardware /Users/lokki/Library/Arduino15/packages -hardware /Users/lokki/Documents/Arduino/hardware -tools /Applications/new arduino/Arduino.app/Contents/Java/tools-builder -tools /Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/lokki/Library/Arduino15/packages -built-in-libraries /Applications/new arduino/Arduino.app/Contents/Java/libraries -libraries /Users/lokki/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path /var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_build_763347 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_modified_sketch_211930/USBH_MIDI_dump.ino
/Applications/new arduino/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/new arduino/Arduino.app/Contents/Java/hardware -hardware /Users/lokki/Library/Arduino15/packages -hardware /Users/lokki/Documents/Arduino/hardware -tools /Applications/new arduino/Arduino.app/Contents/Java/tools-builder -tools /Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/lokki/Library/Arduino15/packages -built-in-libraries /Applications/new arduino/Arduino.app/Contents/Java/libraries -libraries /Users/lokki/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path /var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_build_763347 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_modified_sketch_211930/USBH_MIDI_dump.ino
Using board 'uno' from platform in folder: /Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_build_763347/sketch/USBH_MIDI_dump.ino.cpp" -o "/dev/null"
"/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/lokki/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0" "/var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_build_763347/sketch/USBH_MIDI_dump.ino.cpp" -o "/dev/null"
"/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/lokki/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI/src" "/var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_build_763347/sketch/USBH_MIDI_dump.ino.cpp" -o "/dev/null"
"/Applications/new arduino/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/lokki/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0" "-I/Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/a [... cut ...]
/var/folders/7j/g12t223s4x33pfnq55x9qzkc0000gn/T/arduino_modified_sketch_211930/USBH_MIDI_dump.ino: In function 'void loop()':
USBH_MIDI_dump:26: error: could not convert 'Midi' from 'USBH_MIDI' to 'bool'
   if( Midi ) {
            ^
Using library USB_Host_Shield_Library_2.0 at version 1.3.2 in folder: /Users/lokki/Documents/Arduino/libraries/USB_Host_Shield_Library_2.0 
Using library SPI at version 1.0 in folder: /Applications/new arduino/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
exit status 1
could not convert 'Midi' from 'USBH_MIDI' to 'bool'
``

lokkiikkol, 2019-06-05:
after some more tinkering, this works:

`#include <usbh_midi.h>
#include <usbhub.h>
#include <SPI.h>

USB Usb;
USBHub Hub(&Usb);
USBH_MIDI Midi(&Usb);
uint8_t start = 0;
void setup()
{
    Serial.begin(115200);

    if (Usb.Init() == -1) {
        while (1); //halt
    }//if (Usb.Init() == -1...
    delay( 200 );
}

uint8_t msg2 = 250;
uint8_t msg1 = 248;
uint8_t msg[3];

void loop()
{
  Usb.Task();
    msg[0] = msg1;
    Midi.SendData(msg,1);
    delay(30);
}`

so i guess port number is zero based :-) also, what is the if(Midi) for? is it ok to use the code without it?

YuuichiAkagawa, 2019-06-06:
Oops. It was not included in the release package yet.
https://github.com/felis/USB_Host_Shield_2.0/pull/343

The old implementation is
```
void loop()
{
  Usb.Task();
  if ( Usb.getUsbTaskState() == USB_STATE_RUNNING ) {
    if (!start) {
        msg[0] = msg2;
        Midi.SendData(msg, 1);
        start = 1;
        Usb.Task();
    }
    msg[0] = msg1;
    Midi.SendData(msg, 1);
    delay(30);
  }
}
```

> so i guess port number is zero based :-) 

You right.

YuuichiAkagawa, 2020-07-26:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2020-07-26 by YuuichiAkagawa