YuuichiAkagawa/USBH_MIDI #48: Digitech RP-355, why No MIDI Device?

Rifkat6519, 2019-11-20:
Hi!
Sorry for bad English.
Why library does not see my guitar processor RP-355(composite usb device)? 

1.     I connect  my RP-355 to the PC (usbview.exe) and show(fragment):
=============================
  Interface Descriptor:
  bInterfaceNumber: 0x07                               =7 
  bAlternateSetting: 0x00
  bNumEndpoints: 0x02
  bInterfaceClass: 0x01 (Audio)                      ++ buf_ptr[5] ==USB_CLASS_AUDIO
  bInterfaceSubClass: 0x03 (MIDI Streaming) ++buf_ptr[5] ==USB_CLASS_AUDIO
  bInterfaceProtocol: 0x00
  iInterface: 0x04
  0x0409: "DigiTech RP355 MIDI"
  0x0409: "DigiTech RP355 MIDI"
=============================

2. Next, I start sketch "Usb_Desc", and show: 

===
Start
01
--
String Descriptors:
Manufacturer:		DigiTech
Product:		DigiTech RP355

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		1210
Product ID:		001D
Revision ID:		0001
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01
Total length truncated to 256 bytes
Configuration descriptor:
Total length:		0160
Num.intf:		05
Conf.value:		01
Conf.string:		02
Attr.:			80
Max.pwr:		00

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		0A
Type:		24
Contents:	01000134000201020C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020101010002030000030924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030203060001030C24
Unknown descriptor:
Length:		0C
Type:		24
Contents:	020303060002030000030924
Unknown descriptor:
Length:		09
Type:		24
Contents:	030401010003030904

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	01010101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010202100144AC000905

Endpoint descriptor:
Endpoint address:	01
Attr.:			0D
Max.pkt size:		00C0
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01010100000904

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	01010101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010203180144AC000905

Endpoint descriptor:
Endpoint address:	01
Attr.:			0D
Max.pkt size:		0120
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01010100000904

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	01040101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010202100144AC000905

Endpoint descriptor:
Endpoint address:	82
Attr.:			0D
Max.pkt size:		00C0
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01010100000904

Interface descriptor:
Intf.number:		02
Alt.:			02
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		00
Intf.string:		03
Unknown descriptor:
Length:		07
Type:		24
Contents:	01040101000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010203180144AC000905

Endpoint descriptor:
Endpoint address:	82
Attr.:			0D
Max.pkt size:		0120
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	0100010AE300E7

Addr:1(0.0.1)
=================

3. With enabled Debug:
=====================================
Usb.Init=ok
MIDI Init
Addr:01
VID:1210 PID:001D #Conf:01

Conf:01 Int:00 Alt:00 EPs:00 IntCl:01 IntSubCl:01
No MIDI Device

Conf:01 Int:01 Alt:00 EPs:00 IntCl:01 IntSubCl:02
No MIDI Device

Conf:01 Int:01 Alt:01 EPs:01 IntCl:01 IntSubCl:02
No MIDI Device
-EPAddr:01 bmAttr:0D MaxPktSz:C0

Conf:01 Int:01 Alt:02 EPs:01 IntCl:01 IntSubCl:02
No MIDI Device
-EPAddr:01 bmAttr:0D MaxPktSz:20

Conf:01 Int:02 Alt:00 EPs:00 IntCl:01 IntSubCl:02
No MIDI Device

Conf:01 Int:02 Alt:01 EPs:01 IntCl:01 IntSubCl:02
No MIDI Device
-EPAddr:82 bmAttr:0D MaxPktSz:C0

Conf:01 Int:02 Alt:02 EPs:01 IntCl:01 IntSubCl:02
No MIDI Device
-EPAddr:82 bmAttr:0D MaxPktSz:20

NumEP:01
=====================================
Tell me please, what to do?

YuuichiAkagawa, 2019-11-20:
Oh, Your controller's descriptor is too big.
```
Total length truncated to 256 bytes
Configuration descriptor:
Total length: 0160
```
The current implementation only supports up to 256 bytes.

Try increasing the buffer size to 384 bytes.
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/usbh_midi.h#L34
```
#define DESC_BUFF_SIZE        256
```

Rifkat6519, 2019-11-20:
Thank you very much! After setting #define DESC_BUFF_SIZE 384. RP-355 started changing presets.
[image]

banoydakila, 2021-04-29:
Hi, Rifkat6519 can I please see your code to get ideas? I also want to create a controller for my Nuxmg300 but don't know how to send CC message

Rifkat6519, 2021-04-29:
Hi!

My midi footwitch for guitar processor Digitech RP-355.

	1. Arduino Leonardo r3(china)
	2. USB Host Shield(china)
	3. TM1638 "LED&KEY"(8 leds + 8 Keys)(china)
	4. 8-buttons

Rifkat.

Rifkat6519, 2021-04-29:
https://www.youtube.com/watch?v=JGsWVKbrNZs&t=134s

banoydakila, 2021-05-02:
So my goal is to send cc message from arduino to the device. However this code doesn't work. descriptor is also 256 and I've changed it 384. I'm sending this cc message that I got from midi Ox "B0 3C 01" Can you please help me? skateboard11asd@gmail.com

#include <SPI.h>
#include <usbh_midi.h>
USB Usb;
USBH_MIDI Midi(&Usb);

void sendMidi(uint8_t ch, uint8_t ev, uint8_t val)
{
  Usb.Task();
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING ){
  Serial.println("RUNNING");
  uint8_t buf[3];
  buf[0] = ch; // 0xB0;
  buf[1] = ev; // 0x3C;
  buf[2] = val; // 1;
  const uint8_t test = Midi.SendData(buf);
  Serial.println(test);
  delay(1000);

}
}

void setup() {
  Serial.begin(115200);
  while (!Serial);
  if (Usb.Init() == -1) {
    Serial.println("Usb.Init-Bad!");
    while(1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop() {
sendMidi(0xb0, 0x3c, 1);
}

YuuichiAkagawa, 2021-05-02:
@banoydakila 
Try the develop branch version.
https://github.com/YuuichiAkagawa/USBH_MIDI/tree/develop

And example
https://gist.github.com/YuuichiAkagawa/4226da674be9fc619b67e6607b797608

banoydakila, 2021-05-02:
OMG! this worked like a charm. Thank you very much Sir Yuuichi!

Rifkat6519, 2021-05-02:
HI! 

I replaced 2 files v0.6.0 in my sketch and 
show errors when compiling. 
In Attach my fixed files( __usbh_midi.cpp and  __usbh_midi.h)
Compare this files with your last version files(*.h, *.cpp). 

log compile sketch:

libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::Release()'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::USBH_MIDI(USB*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::USBH_MIDI(USB*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::parseConfigDescr(unsigned char, unsigned char)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::setupDeviceSpecific()'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::Init(unsigned char, unsigned char, bool)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::RecvData(unsigned int*, unsigned char*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::lookupMsgSize(unsigned char, unsigned char)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::RecvData(unsigned char*, bool)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::RecvRawData(unsigned char*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::countSysExDataSize(unsigned char*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::SendSysEx(unsigned char*, unsigned int, unsigned char)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::SendData(unsigned char*, unsigned char)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::SendRawData(unsigned int, unsigned char*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::extractSysExData(unsigned char*, unsigned char*)'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::epDataOutIndexVSP'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
(.text+0x0): multiple definition of `USBH_MIDI::epDataInIndexVSP'
libraries\USB_Host_Shield_2.0-master\MY_usbh_midi.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries\USB_Host_Shield_2.0-master\usbh_midi.cpp.o (symbol from plugin): In function `USBH_MIDI::Release()':
[... cut ...]
        uint8_t SendRawData(uint16_t bytes_send, uint8_t *dataptr);
        // backward compatibility functions
        inline uint8_t RcvData(uint16_t *bytes_rcvd, uint8_t *dataptr) { return RecvData(bytes_rcvd, dataptr); };
        inline uint8_t RcvData(uint8_t *outBuf) { return RecvData(outBuf); };

        // USBDeviceConfig implementation
        virtual uint8_t Init(uint8_t parent, uint8_t port, bool lowspeed);
        virtual uint8_t Release();
        virtual uint8_t GetAddress() { return bAddress; };
};
#endif //_USBH_MIDI_H_

YuuichiAkagawa, 2021-05-02:
Please remove MY_usbh_midi.h and MY_usbh_midi.cpp.
Or move it from the libraries directory.

Rifkat6519, 2021-05-02:
Yes you are right. Everything works now. Thank you!

От: Yuuichi Akagawa
Отправлено: 2 мая 2021 г. в 11:14
Кому: YuuichiAkagawa/USBH_MIDI
Копия: Rifkat6519; Author
Тема: Re: [YuuichiAkagawa/USBH_MIDI] Digitech RP-355, why No MIDI Device?(#48)

Please remove MY_usbh_midi.h and MY_usbh_midi.cpp.
Or move it from the libraries directory.

banoydakila, 2021-05-04:
Hi sir Yuuichi! I just want to ask if how can I retrieve or get cc data from the device and store then on a variable. I'm messing with the RecvData but it keeps on printing "0". Thank you

YuuichiAkagawa, 2021-05-04:
@banoydakila 
This issue is about RP-355.
Pleas open new discussion or issue.

YuuichiAkagawa, 2021-05-09:
Fixed in [v0.6.0](https://github.com/YuuichiAkagawa/USBH_MIDI/releases/tag/0.6.0).  
Reference: #72

closed 2021-05-09 by YuuichiAkagawa