YuuichiAkagawa/USBH_MIDI #49: Why module "LED&KEY"(TM1638) not working?

Rifkat6519, 2019-11-22:
Hi!
Sorry for my bad english!
I want to use the “LED&KEY“ module on the TM1638 chip to switch presets for my “Digitech RP-355“ guitar processor via USBMIDI. For testing, I use example ”tm1638_one_module_example.ino” from examples library TM1638 (https://github.com/rjbatista/tm1638-library).
To sketch I add the next lines:

#include <usbh_midi.h>
#include <usbhub.h>
USB Usb;
USBH_MIDI Midi(&Usb);

But when starting, the LED indicators do not light up and the buttons do not respond to pressing.
I found, that if in setup() I comment line with “Usb.Init ()“
// if (Usb.Init() == -1)  while (1);
then the “LED & KEY” module and the sketch begin to work.

Thank you in advance for your help!

`/*
Library examples for TM1638.

Copyright (C) 2011 Ricardo Batista <rjbatista at gmail dot com>

This program is free software: you can redistribute it and/or modify
it under the terms of the version 3 GNU General Public License as
published by the Free Software Foundation.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include <usbh_midi.h>
#include <usbhub.h>

#include <TM1638.h>

// define a module on data pin 8, clock pin 9 and strobe pin 7
TM1638 module(8, 9, 7);

USB Usb;
USBH_MIDI Midi(&Usb);

void setup() {
   //if (Usb.Init() == -1)  while (1);
  // display a hexadecimal number and set the left 4 dots
  module.setDisplayToHexNumber(0x1234ABCD, 0xF0);
}

void loop() {
  byte keys = module.getButtons();

  // light the first 4 red LEDs and the last 4 green LEDs as the buttons are pressed
  module.setLEDs(((keys & 0xF0) << 8) | (keys & 0xF));
}`

YuuichiAkagawa, 2019-11-22:
Which USB Host Shield are you using?

Rifkat6519, 2019-11-22:
https://ru.aliexpress.com/item/1638734824.html?spm=a2g0s.9042311.0.0.274233ed8IPudC

Rifkat6519, 2019-11-22:
and https://ru.aliexpress.com/item/32672044070.html?spm=a2g0s.9042311.0.0.274233ed8IPudC

Rifkat6519, 2019-11-22:
and https://ru.aliexpress.com/item/32923016834.html?spm=a2g0s.9042311.0.0.274233ed8IPudC

YuuichiAkagawa, 2019-11-23:
The D9 pin is connect to the INT of USB Host Shield.
Do not use D9.

Rifkat6519, 2019-11-23:
Yes you are right. I changed pin 9 to 6 and my module "LED&KEY" began to work.
Thanks!

YuuichiAkagawa, 2019-11-23:
😄

closed 2019-11-23 by YuuichiAkagawa