YuuichiAkagawa/USBH_MIDI #50: Problem Send SysEx message

Rifkat6519, 2019-12-07:
Hi, Yuuichi!

Question:
If send program change(2 byte message) to  my device over usb it stability change preset,
but if i send SysEx message(15 byte), my device does not receive a command stably.
in monitor I show "rc=4" as result Midi.SendData(dataptr, 0);

=============================================================
// Test , 15 byte SysEx message Effect "Compressor(On/Off)"
=============================================================
uint8_t COMP_OFF[] = {0xf0, 0x00, 0x00, 0x10, 0x00, 0x5e, 0x09, 0x41, 0x20, 0x00, 0x41, 0x04, 0x00, 0x63, 0xf7};
uint8_t COMP_ON[]  = {0xf0, 0x00, 0x00, 0x10, 0x00, 0x5e, 0x09, 0x41, 0x20, 0x00, 0x41, 0x04, 0x01, 0x62, 0xf7};
bool enabled=false;

void loop() {
    checkMyButtons();
    Usb.Task();
    if ( Midi ) { // USB
       if (pressedButton] != 1) 
          if (enabled) Send_SysEx(COMP_OFF, sizeof(COMP_OFF));
          else Send_SysEx(COMP_ON, sizeof(COMP_ON));
          enabled=(!enabled);
       }
   }
}

void Send_SysEx(uint8_t *dataptr, uint16_t datasize) {
  Usb.Task();
  //uint8_t rc = Midi.SendSysEx(dataptr, datasize, 0);
  uint8_t rc = Midi.SendData(dataptr, 0);
#if defined(UseDebug)
  Serial.print(F("Send_SysEx():"));
  Serial.print(rc, HEX);
  prnarr(":", dataptr, datasize, HEX);
#endif //defined(UseDebug)    
}
=============================================================
what can be done?

my config: 
Arduino Leonardo + USB Host Shield + my guitar processor "Digitech RP-355" midi over USB.

I am looking examples where arduino sends and receive a SySex messages to a midi device via USB as dialog. Can you show some example of sketch?

thanks.

YuuichiAkagawa, 2019-12-07:
The SysEx example is here.
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/eVY1_sample/eVY1_sample.ino

"rc=4" is NAK. It is meaning 'Data packet not accepted'.
https://github.com/felis/USB_Host_Shield_2.0/blob/master/max3421e.h#L213-L229

YuuichiAkagawa, 2020-07-26:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2020-07-26 by YuuichiAkagawa