YuuichiAkagawa/USBH_MIDI #51: Using MIDI Converter with nanoKEY Studio on Arduino Pro Mini / mini USB Host Shield

npyskm, 2020-01-17:
当方、ハードウェアソフトウェア共に初心者です。

こちらの「USBH_MIDI」をArduino IDEに導入した際にサンプルコードに含まれる「USBH_MIDI_converter」を、手持ちだった「Arduino Pro Mini 328 3.3V 8MHz ＋ ミニUSBホストシールド (＋ MIDIアウト回路)」上で動かすことで、「KORG nanoKEY Studio」をUSBキーボードとして使えないかと考えています。標準の「Arduino ＋ USBホストシールド」でないことと、およびここに書かれている対応機種の後継機種であることが、この意図に対してどのように影響するのかわからず、質問をします。

[質問]
nanoKEY無印は対応機種に含まれているものの、nanoKEY Studioが現状では含まれていないことは理解しています。そのそもそもの意味としては、ハードウェアの何らかのIDの制限などの理由で、対応機種に含まれていないものは全く動かないという話なのでしょうか、それともソフトウェアを多少触ることによって動くようになる感じでしょうか、またはおおかたは動くはずなのでしょうか。

[試したこと]
1)「USBH_MIDI_converter」に、LED点灯を使った簡易デバッグをすることで、Usb.Init()は通過することはわかった(これにより、USBH_MIDI_converterはミニUSBホストシールドでも使える、と理解)
2)しかし同様に簡易デバッグによると、Midi.RecvData(outBuf)は出ていない様子。
3)nanoKEY StudioからMIDIとして信号を受け取れていないのかと思い「USBH_MIDI_dump」を書き込んだが、ボーレートを115200とした「シリアルモニタ」窓を開いても何も表示されなかった(使い方が合っているか自信無し)。

何か示唆するにしてもこのような情報が足りない、このようなことを試すべきだ、そもそも用途が違う、動くはずだから故障ではないか、などご指摘ご教示いただけるとありがたいです。

YuuichiAkagawa, 2020-01-17:
Pro miniでも問題なく動作します。
まずは、USB機器として認識できているかどうかを確認する必要があるため、USB Host Shield Library2.0のUSB_descを使って確認をお願いします。出力結果をお知らせください。

nanoKeyでの実行例です。
https://gist.github.com/YuuichiAkagawa/6d1a08b675693e56dac39eb3776a8ab2

また、割と良くあるのがMIDI機器への電源供給が不足しているケースです。nanoKEY Studioは電池駆動もできる様なので、USBからの給電ではなく、電池を使用したセルフパワー動作も確認いただければと思います。

npyskm, 2020-01-20:
USB_descを書き込んでも、起動時の「Start」は「シリアルモニタ」に表示されたものの、nanoKEY Studioを挿しても何も出力されませんでした。ミニUSBホストシールドの初期不良なのかもしれません…(5V確保のためのパターンカットやピンヘッダ付けなどをしているため、返品不可)  買い直しを検討しています。

なお、nanoKEY Studioは電池で動かす場合にはBluetooth専用となり、USB側はもともと使えない仕様のようです。

YuuichiAkagawa, 2020-01-20:
USBマウス等でもUSB_descの結果が得られなければ、USBHostシールドの問題の可能性が高いです。

npyskm, 2020-01-21:
アドバイスたいへんありがとうございます、USBマウスとUSBキーボード(QWERTY)を接続しましたが、シリアルモニタには何も追記されませんでした。ミニUSBホストシールドを再発注していますので、届きましたらまたUSB_descやUSBH_MIDI_dumpなどを試してみます。

YuuichiAkagawa, 2020-01-21:
念のため確認ですが、Arduino Pro miniは 328 3.3V 8MHz版でしょうか？
そして、Arduino IDEでも
- Board: Arduino Pro or Pro mini
- Processor: ATMega328P (3,3V, 8MHz)

を選択していますでしょうか。

USBHostシールドのチェックは、board_qcでできますのでご確認ください。
 SPI long test passedが出力されれば動作しています。(GPIO testは失敗します)

npyskm, 2020-01-23:
> 念のため確認ですが、Arduino Pro miniは 328 3.3V 8MHz版でしょうか？
> そして、Arduino IDEでも

これらに関しては、以前に注文した商品名も一字一句それでしたし、もちろんです、と返答しかけて、念の為わたしが思い付く方法にてチェックしたところ、"Arduino Pro mini 328 3.3V 16MHz"とでも言うべきものなのかな?と言う疑念が沸いております。すみません、ボードが間違っているかもということは、予想しておりませんでした。

[現Arduinoボードについて]
1)"Pro mini ／ 3.3V 8MHz"で設定したArduino IDEのサンプルスケッチ"Blink"にて実測したところ、点灯間隔が半分になっているように見えた(もしや16MHzでは?という疑念)
2)当初から3.3V版だと思い込んでいたのでRAWには5Vを入力しており、またVccに3.3V前後が測定されていることを確認済み(3.3V版で合っているという判断)、またFTDIシリアルUSBは3.3Vで作動させている(3.3V版で合っているという判断)。これらの判断、およびチップに書いてあるATmega328Pという文字と許容動作周波数から、8MHzで動作していると判断した(が、間違っていたかもしれない)。クリスタルの表面の文字は読めず、また周波数が測定できる機材は持っていない。

こちら、ボードは手持ちの1枚だけだったためこれ以上の検証はできず、泣く泣く別の業者から"Arduino Pro mini 328 3.3V 8MHz"も再購入します。また、疑念があるままトラブルシュートするのはセオリから外れると思いつつ、現環境にてご提示の"board_qc"も試してみました。

[現ミニUSBホストシールドについて]
1)"Pro mini ／ 3.3V 8MHz"で設定したArduino IDEのサンプルスケッチ"board_qc"を実行したところ、SPI long test passedが出力(GPIO testはNG)。
2)"3.3V 16MHz"という設定は存在しなかったため、"Pro mini ／ 328 5V 16MHz"で設定したArduino IDEのサンプルスケッチ"board_qc"を実行したところ、同様にSPI long test passedが出力(GPIO testはNG)。

意外だったのですが、IDE上8MHz想定／16MHz想定のどちらでコンパイルされた実行ファイルでも、現ミニUSBホストシールドは正常と出てしまいました。それでも手持ちの低消費電力のUSB機器3機種が無反応だということから、素人考えながら、ミニUSBホストシールドのメインのチップから外部のUSBコネクタまでの間に不具合があるのかなと思っています。

いずれにせよ、USBH_MIDIに対するというよりも大きな疑念が出てきてしまっており、発注済みのミニUSBホストシールドと、Pro mini 328 3.3V 8MHzとが届きましたら、また確認します。

YuuichiAkagawa, 2020-01-23:
詳細なレポートありがとうございます。
mini版は情報が少なく、玄人向けだと思ってます。
もう一つ思い当たる点として、電源は以下のように接続されていますか？
https://twitter.com/YuuichiAkagawa/status/307151034839613441

npyskm, 2020-01-23:
いろいろなことを指摘していただき、ありがとうございます。

> もう一つ思い当たる点として、電源は以下のように接続されていますか？

はい、写真の基板と少しだけ様子が違いますが、わたしの現ミニUSBホストシールドには最初から、基板上のUSB-Aホストコネクタの電源線にのみ5Vが出力され、その他の電源バスはVcc 3.3Vが供給されているままになるよう、パターンカットとジャンパ処理を施しておりました。よって、現ミニUSBホストシールドの電源まわりで不具合を誘発したことはないと考えています。

npyskm, 2020-02-07:
ミニUSBホストシールドはまだなのですが(春節やコロナヴィルスの関係かもしれません)、後に発注したPro mini 328 3.3V 8MHzの方が先に届いてしまいました。とりいそぎLEDやシリアルにてテストしたところ、前のものの動作と違い、正しく8MHzで動いているらしいことが間接的にわかりました。

疑わしいミニUSBホストシールドと繋いでみたところ、状況は(16MHz? 8MHz? に関する部分以外は)同じでした。引き続き新ミニUSBホストシールドの到着を待ちたいと思います。

YuuichiAkagawa, 2021-05-07:
一旦クローズします。必要でしたら再オープンしてください。

closed 2021-05-07 by YuuichiAkagawa