YuuichiAkagawa/USBH_MIDI #52: Problem with Digital Piano

wibicucu, 2020-07-10:
I have MIDI project using this library, I am using Yamaha Clavinova and Roland FP30 for testing, and it is working.
But it is not working With Yamaha P121.  I got value 160 for Usb.getUsbTaskState()  (normally with other 2 piano I got 144)

I try to use USB_desc example project, and found lines that suspicious, it said: 
Total length truncated to 256 bytes

I open usbh_mid.h file and set DESC_BUFF_SIZE from 256 to 512

And yes, it working, I can connect, and send midi notes on/off successfully to Yamaha P121. But I can't receive midi data from the piano.  So I try to change value of MIDI_EVENT_PACKET_SIZE from 64 to 128. and it working, Now I can receive midi data from P121.  This setting also working on Yamaha Clavinova and Roland FP30.

Is it safe to change that value? or will it make the code broken on other midi events?

=============================

YuuichiAkagawa, 2020-07-12:
Please run https://gist.github.com/YuuichiAkagawa/e9274a452aec7733cf22bab8f4f78390

> I open usbh_mid.h file and set DESC_BUFF_SIZE from 256 to 512
Yes.

> MIDI_EVENT_PACKET_SIZE from 64 to 128
Hmm. This is unexpected.

wibicucu, 2020-07-15:
Sorry for MIDI_EVENT_PACKET_SIZE, it seems I'm too fast for conclusion, Cant receive midi data seems intermittent problem regardless to my configuration of MIDI_EVENT_PACKET_SIZE.  I need to turn on device with this sequence to avoid problem
1. Turn on Piano first, waiting several seconds
2. Turn on arduino devices
I do test for several times, and it is stable, midi receive data working normal. (restarting arduino on this state may cause problem occurs again).

=======================
USB_desc.ino output
=======================
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		0499
Product ID:		1718
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		012F
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		01
Unknown descriptor:
Length:		08
Type:		0B
Contents:	000301002000

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_CONTROL>>>
Unknown descriptor:
Length:		09
Type:		24
Contents:	010002085D0000
Unknown descriptor:
Length:		08
Type:		24
Contents:	0A0101030000
Unknown descriptor:
Length:		12
Type:		24
Contents:	0606020C000000000000000000000000
Unknown descriptor:
Length:		11
Type:		24
Contents:	020201010001020300000000000000
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03031707000601000000
Unknown descriptor:
Length:		11
Type:		24
Contents:	020417070001020300000000000000
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03050101000401000000

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>
Unknown descriptor:
Length:		10
Type:		24
Contents:	0102000101000000020300000000
Unknown descriptor:
Length:		06
Type:		24
Contents:	02010210

Endpoint descriptor:
Endpoint address:	07
Attr.:			05
Max.pkt size:		00B4
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	010000000000

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>
Unknown descriptor:
Length:		10
Type:		24
Contents:	0105000101000000020300000000
Unknown descriptor:
Length:		06
Type:		24
Contents:	02010210

Endpoint descriptor:
Endpoint address:	86
Attr.:			25
Max.pkt size:		00B4
Polling interval:	01
Unknown descriptor:
Length:		08
Type:		25
Contents:	010000000000

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00

<<<MIDIStreaming>>>

MS Interface descriptor
07240100014100
DescriptorSubtype: 01
CS_INTERFACE :	MS_HEADER
MIDIStreaming SubClass Specification Release number: 0100
wTotalLength:		0041

MS Interface descriptor
062402020100
DescriptorSubtype: 02
CS_INTERFACE :	MIDI_IN_JACK(External)
JackType:	02
JackID:		01

MS Interface descriptor
092403016101010100
DescriptorSubtype: 03
CS_INTERFACE :	MIDI_OUT_JACK(Embedded)
JackType:	01
JackID:		61
NrInputPins:	01
010100

MS Interface descriptor
062402012100
DescriptorSubtype: 02
CS_INTERFACE :	MIDI_IN_JACK(Embedded)
JackType:	01
JackID:		21

MS Interface descriptor
092403024101210100
DescriptorSubtype: 03
CS_INTERFACE :	MIDI_OUT_JACK(External)
JackType:	02
JackID:		41
NrInputPins:	01
210100

Endpoint descriptor:
Endpoint address:	03
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

MS Endpoint descriptor
0525010121
CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 01
21

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

MS Endpoint descriptor
0525010161
CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 01
61

Addr:1(0.0.1)

YuuichiAkagawa, 2020-07-15:
Thank you for your report.
It is difficult to determine when self-powered devices will be available.

YuuichiAkagawa, 2020-07-26:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2020-07-26 by YuuichiAkagawa