YuuichiAkagawa/USBH_MIDI #53: Running Status - Is there a switch to enable ?

ADI8421, 2020-07-15:
Is there a switch somewhere (true/false etc) to enable Running Status ? Many thanks

YuuichiAkagawa, 2020-07-15:
What is switch? Please tell me how to use it.

ADI8421, 2020-07-15:
My question refers to Midi output 'Running Status' 
The firmware reads 'midi running status' input great. The problem is on the midi output. Running status is disabled, which means every midi message has the status byte included. This just increases the bandwidth requirement, especially with Pitch Wheel and Mod Wheel data.
So my question is: Can 'running status' be enabled with a simple change in the source code ? Many thanks.

ADI8421, 2020-07-16:
It's OK.  I wrote some code inside the USB_MIDI_converter.ino sketch which takes care of running status problem.
    do {
      if( (size=Midi.RcvData(outBuf)) > 0 ){

// Code goes here to enable Running Status.

        //MIDI Output
        _MIDI_SERIAL_PORT.write(outBuf, size);
      }
    }while(size>0);

YuuichiAkagawa, 2020-07-16:
Thank you, I understand.
There is no function to handle'Running Status' in this library.
You have to implement it yourself.

closed 2020-07-26 by YuuichiAkagawa