YuuichiAkagawa/USBH_MIDI #55: LaunchKey Mini InControl Mode

lapinuelle, 2020-08-06:
Hello, sir! Is there any way to activate InControl Mode on LaunchKey Mini as described here: 
https://www.partsnotincluded.com/how-to-control-the-leds-on-a-novation-launchkey-mini-ii/

but using your library?

YuuichiAkagawa, 2020-08-08:
Please run https://gist.github.com/YuuichiAkagawa/e9274a452aec7733cf22bab8f4f78390

lapinuelle, 2020-08-08:
13:38:58.456 -> Start
13:38:59.225 -> 
13:38:59.225 -> 
13:38:59.225 -> 01
13:38:59.225 -> --
13:38:59.225 -> 
13:38:59.225 -> Device descriptor: 
13:38:59.225 -> Descriptor Length:	12
13:38:59.225 -> Descriptor type:	01
13:38:59.225 -> USB version:		0110
13:38:59.225 -> Device class:		00
13:38:59.225 -> Device Subclass:	00
13:38:59.225 -> Device Protocol:	00
13:38:59.225 -> Max.packet size:	40
13:38:59.225 -> Vendor  ID:		1235
13:38:59.225 -> Product ID:		0035
13:38:59.225 -> Revision ID:		0000
13:38:59.225 -> Mfg.string index:	01
13:38:59.225 -> Prod.string index:	02
13:38:59.225 -> Serial number index:	00
13:38:59.225 -> Number of conf.:	01
13:38:59.225 -> 
13:38:59.225 -> Configuration descriptor:
13:38:59.225 -> Total length:		0063
13:38:59.225 -> Num.intf:		02
13:38:59.225 -> Conf.value:		01
13:38:59.225 -> Conf.string:		00
13:38:59.225 -> Attr.:			80
13:38:59.225 -> Max.pwr:		32
13:38:59.225 -> 
13:38:59.225 -> Interface descriptor:
13:38:59.225 -> Intf.number:		00
13:38:59.225 -> Alt.:			00
13:38:59.225 -> Endpoints:		00
13:38:59.225 -> Intf. Class:		01
13:38:59.225 -> Intf. Subclass:		01
13:38:59.225 -> Intf. Protocol:		00
13:38:59.225 -> Intf.string:		03
13:38:59.225 -> 
13:38:59.225 -> <<<AUDIO_CONTROL>>>
13:38:59.225 -> Unknown descriptor:
13:38:59.225 -> Length:		09
13:38:59.225 -> Type:		24
13:38:59.225 -> Contents:	01000109000101
13:38:59.225 -> 
13:38:59.225 -> Interface descriptor:
13:38:59.225 -> Intf.number:		01
13:38:59.225 -> Alt.:			00
13:38:59.225 -> Endpoints:		02
13:38:59.225 -> Intf. Class:		01
13:38:59.225 -> Intf. Subclass:		03
13:38:59.225 -> Intf. Protocol:		00
13:38:59.225 -> Intf.string:		03
13:38:59.225 -> 
13:38:59.225 -> <<<MIDIStreaming>>>
13:38:59.225 -> 
13:38:59.225 -> MS Interface descriptor
13:38:59.225 -> 07240100013F00
13:38:59.225 -> DescriptorSubtype: 01
13:38:59.225 -> CS_INTERFACE :	MS_HEADER
13:38:59.225 -> MIDIStreaming SubClass Specification Release number: 0100
13:38:59.259 -> wTotalLength:		003F
13:38:59.259 -> 
13:38:59.259 -> MS Interface descriptor
13:38:59.259 -> 062402010104
13:38:59.259 -> DescriptorSubtype: 02
13:38:59.259 -> CS_INTERFACE :	MIDI_IN_JACK(Embedded)
13:38:59.259 -> JackType:	01
13:38:59.259 -> JackID:		01
13:38:59.259 -> 
13:38:59.259 -> MS Interface descriptor
13:38:59.259 -> 062402010405
13:38:59.259 -> DescriptorSubtype: 02
13:38:59.259 -> CS_INTERFACE :	MIDI_IN_JACK(Embedded)
13:38:59.259 -> JackType:	01
13:38:59.259 -> JackID:		04
13:38:59.259 -> 
13:38:59.259 -> MS Interface descriptor
13:38:59.259 -> 092403010201010104
13:38:59.259 -> DescriptorSubtype: 03
13:38:59.259 -> CS_INTERFACE :	MIDI_OUT_JACK(Embedded)
13:38:59.259 -> JackType:	01
13:38:59.259 -> JackID:		02
13:38:59.259 -> NrInputPins:	01
13:38:59.259 -> 010104
13:38:59.259 -> 
13:38:59.259 -> MS Interface descriptor
13:38:59.259 -> 092403010301020205
13:38:59.259 -> DescriptorSubtype: 03
13:38:59.259 -> CS_INTERFACE :	MIDI_OUT_JACK(Embedded)
13:38:59.259 -> JackType:	01
13:38:59.259 -> JackID:		03
13:38:59.259 -> NrInputPins:	01
13:38:59.259 -> 020205
13:38:59.259 -> 
13:38:59.259 -> Endpoint descriptor:
13:38:59.259 -> Endpoint address:	81
13:38:59.259 -> Attr.:			03
13:38:59.259 -> Max.pkt size:		0040
13:38:59.259 -> Polling interval:	01
13:38:59.259 -> 
13:38:59.259 -> MS Endpoint descriptor
13:38:59.259 -> 062501020203
13:38:59.259 -> CS_ENDPOINT :	MS_GENERAL
13:38:59.259 -> bNumEmbMIDIJack: 02
13:38:59.259 -> 0203
13:38:59.259 -> 
13:38:59.259 -> Endpoint descriptor:
13:38:59.259 -> Endpoint address:	02
13:38:59.259 -> Attr.:			03
13:38:59.259 -> Max.pkt size:		0040
13:38:59.259 -> Polling interval:	01
13:38:59.259 -> 
13:38:59.259 -> MS Endpoint descriptor
13:38:59.259 -> 062501020104
13:38:59.259 -> CS_ENDPOINT :	MS_GENERAL
13:38:59.259 -> bNumEmbMIDIJack: 02
13:38:59.259 -> 0104
13:38:59.259 -> 
13:38:59.259 -> 
13:38:59.259 -> Addr:1(0.0.1)

YuuichiAkagawa, 2020-08-08:
You can probably switch to InControl mode by the following code.
```
//InControl mode on
void inControlOn()
{
  uint8_t buf[3];
  buf[0] = 0x90;
  buf[1] = 0x0c;
  buf[2] = 0x7f;
  Midi.SendData(buf, 1);
}

//InControl mode off
void inControlOff()
{
  uint8_t buf[3];
  buf[0] = 0x90;
  buf[1] = 0x0c;
  buf[2] = 0x00;
  Midi.SendData(buf, 1);
}
```

lapinuelle, 2020-08-08:
Worked like a charm! Big thanks to you!

YuuichiAkagawa, 2020-08-08:
:smile:

closed 2020-08-08 by YuuichiAkagawa