YuuichiAkagawa/USBH_MIDI #58: Problem midi send with PSR SX900

wibicucu, 2020-11-04:
I built midi device using this library and already tested and running well on veriety of keyboard instrument. But it have problems on keyboard yamaha  PSR SX900 (it works well on PSR SX600)

Midi receive data works just fine, But midi send is inconsistent.

So I back to very basic samples EVY1_sample.ino , and the result sounding notes are not consistent, there are some notes event skipped to note on / note off.

I try to capture error code on this line
errcode=Midi.SendData(buf);

and errorcode not always return value 0, sometimes it return value 4, 

I already upgrde firmware of keyboard SX900 device to the latest firmware available (1.05) , testing Midi send / receive data SX900 to computer also work fine.

Hope you can help me to figure out whats wrong.

Thanks.

This is output of midi_desc.ino
================================
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0210
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		0499
Product ID:		1618
Revision ID:		0314
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0075
Num.intf:		02
Conf.value:		01
Conf.string:		04
Attr.:			C0
Max.pwr:		01

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		00
Intf.string:		00

<<<AUDIO_CONTROL>>>
Unknown descriptor:
Length:		09
Type:		24
Contents:	01000109000101

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00

<<<MIDIStreaming>>>

MS Interface descriptor
07240100015100
DescriptorSubtype: 01
CS_INTERFACE :	MS_HEADER
MIDIStreaming SubClass Specification Release number: 0100
wTotalLength:		0051

MS Interface descriptor
062402020100
DescriptorSubtype: 02
CS_INTERFACE :	MIDI_IN_JACK(External)
JackType:	02
JackID:		01

MS Interface descriptor
092403010201010100
DescriptorSubtype: 03
CS_INTERFACE :	MIDI_OUT_JACK(Embedded)
JackType:	01
JackID:		02
NrInputPins:	01
010100

MS Interface descriptor
062402010300
DescriptorSubtype: 02
CS_INTERFACE :	MIDI_IN_JACK(Embedded)
JackType:	01
JackID:		03

MS Interface descriptor
092403020401030100
DescriptorSubtype: 03
CS_INTERFACE :	MIDI_OUT_JACK(External)
JackType:	02
JackID:		04
NrInputPins:	01
030100

MS Interface descriptor
062402010500
DescriptorSubtype: 02
CS_INTERFACE :	MIDI_IN_JACK(Embedded)
JackType:	01
JackID:		05

MS Interface descriptor
092403020601050100
DescriptorSubtype: 03
CS_INTERFACE :	MIDI_OUT_JACK(External)
JackType:	02
JackID:		06
NrInputPins:	01
050100

Endpoint descriptor:
Endpoint address:	01
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

MS Endpoint descriptor
062501020305
CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 02
0305

Endpoint descriptor:
Endpoint address:	81
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

MS Endpoint descriptor
0525010102
CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 01
02

Addr:1(0.0.1)

YuuichiAkagawa, 2020-11-05:
I read the [PSR-SX900/PSR-SX700 reference manual](https://europe.yamaha.com/files/download/other_assets/7/1279207/psrsx900_sx700_en_rm_a0.pdf).
P.124 MIDI Receive Channel Settings

Is the Receiver monitor unresponsive?

wibicucu, 2020-11-06:
Midi receiver monitor also inconsistent
When I send midi data (note on/note off)
- sometimes it blinking and the keyboard do the command (sound correctly)
- sometimes it blinking but the keyboard do nothing
- sometimes it not blinking and keyboard do nothing

YuuichiAkagawa, 2020-11-09:
Hmm.
> errcode=Midi.SendData(buf);
> and errorcode not always return value 0, sometimes it return value 4,

The return value 4 is "NAK from peripheral". It means that PSR-SX900 cannot be received now.

wibicucu, 2020-11-11:
Thanks for your information,  because the error code means that PSR-SX900 cannot be received "now". so maybe the keyboard is in a middle of doing something that can't be interrupted,  I think it should be retry later,  so I create a loop to retry midi send command if response code is 4. and it works without adding significant latency because usually it works on 2nd retries.

YuuichiAkagawa, 2020-11-23:
If you don't mind, please share the device information on Wiki.
https://github.com/YuuichiAkagawa/USBH_MIDI/wiki

closed 2020-11-23 by YuuichiAkagawa