YuuichiAkagawa/USBH_MIDI #59: USBMIDI bridge

teatralnik, 2020-11-15:
Hello. Thank you for your library.
Is it possible to make a bridge between 2 usb-midi devices using your library?

YuuichiAkagawa, 2020-11-15:
This library can handle multiple devices.
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USB_MIDI_converter_multi/USB_MIDI_converter_multi.ino

However, there is no way to identify each device.
(You can get USB Vendor ID and Product ID)

YuuichiAkagawa, 2020-11-19:
How is it afterwards?

I made an example.
```
//********************************************************************************
// Example of bridging two devices (One-way)
//  Device1 : KORG nanoKEY VID=0x0944, PID=0x010d 
//  Device1 : eVY1 Shield  VID=0x2938, PID=0x0102
// *******************************************************************************
#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

//KORG nanoKEY
#define MIDI1_VID (0x0944)
#define MIDI1_PID (0x010d)
//eVY1 Shield
#define MIDI2_VID (0x2938)
#define MIDI2_PID (0x0102)

USB Usb;
USBHub  Hub1(&Usb);
USBH_MIDI Midi1(&Usb);
USBH_MIDI Midi2(&Usb);
USBH_MIDI *Midi[] {&Midi1, &Midi2};

uint16_t pid[2], vid[2];
uint8_t idx1st, idx2nd;

void setup()
{
  pid[0] = pid[1] = 0;
  vid[0] = vid[1] = 0;
  idx1st = idx2nd = 0xff;

  Serial.begin(115200);

  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );

  Serial.println("\r\nConnecting...");

  //wait for connect
  uint8_t idx = 0;
  while (1) {
    Usb.Task();

    if (*Midi[idx]) {
      // Identify the USB MIDI device
      MIDI_identify(idx);
    }
    if ( idx1st != 0xff && idx2nd != 0xff) {
      break;
    }
    delay(10);
    idx = (idx == 1) ? 0 : 1;
  }

  Serial.println("done.");
}

void loop()
{
  Usb.Task();
  uint8_t size;
  uint8_t outBuf[3];
  uint32_t t1 = (uint32_t)micros();
  //One-way brdige
  do {
    if ( (size = Midi[idx1st]->RecvData(outBuf)) > 0 ) {
      Midi[idx2nd]->SendData(outBuf);
    }
  } while (size > 0);
  //delay(1ms) if you need.
  doDelay(t1, (uint32_t)micros(), 1000);
}

// Identify the USB MIDI device
void MIDI_identify(uint8_t idx)
{
  //first call?
  if (Midi[idx]->idVendor() != vid[idx] || Midi[idx]->idProduct() != pid[idx]) {
    vid[idx] = Midi[idx]->idVendor();
    pid[idx] = Midi[idx]->idProduct();
    Serial.print(" Configured: ");
    Serial.print(idx);
    Serial.print(",");
    Serial.print(vid[idx], HEX);
    Serial.print(",");
    Serial.print(pid[idx], HEX);
  }
  //Identify the first device
  if ( vid[idx] == MIDI1_VID && pid[idx] == MIDI1_PID && idx1st == 0xff) {
    idx1st = idx;
    Serial.println(" - 1st device");
    return;
  }
  //Identify the second device
  if ( vid[idx] == MIDI2_VID && pid[idx] == MIDI2_PID && idx2nd == 0xff) {
    idx2nd = idx;
    Serial.println(" - 2nd device");
    return;
  }
}

// Delay time (max 16383 us)
void doDelay(uint32_t t1, uint32_t t2, uint32_t delayTime)
{
  uint32_t t3;

  t3 = t2 - t1;
  if ( t3 < delayTime ) {
    delayMicroseconds(delayTime - t3);
  }
}
```

teatralnik, 2020-11-20:
Big thanks!!!!I've ordered usb host shield, i'll test it immediately as soon as it arrives and leave a reply.

YuuichiAkagawa, 2020-11-23:
Each device is identified by USB Vendor ID (VID) and Product ID (PID) in this example.  
These values can be found by running examples/USB_MIDI_desc.

YuuichiAkagawa, 2021-05-09:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2021-05-09 by YuuichiAkagawa