YuuichiAkagawa/USBH_MIDI #60: Message-skipping issue, on SAMD21 boards

Kuba0040, 2020-12-12:
Hello,
Thank you so much for making this library, however I ran into a issue while using it on a SAMD21 based microcontroller (Seeeduino XIAO). Mainly, every once in a while when playing on a keyboard being read by the microcontroller, sometimes the microcontroller fails to catch Note Off messages, meaning that the key you released will keep on playing until you press, and release it again. The microcontroller I am using (Seeeduino XIAO), while it's not mentioned officially to be supported by Your library, runs on the SAMD21 M0 microcontroller, the same one found in the Arduino Zero, at the same clock speed.

I have tried some things, which seemed to make it skip messages less often, like removing the USBHub library from Your code as well as removing the "**if (Midi)**" line from the main code loop, and just making it run **MIDI_poll()** all the time _(I am not entirely certain about this one, however it seemed to have improved it slightly)_ . However outside of these two discoveries I have failed to eliminate the message skipping problem.

Is there something I am doing wrong that's making the microcontroller misbehave like that? If so please let me know, 
Thank You.

YuuichiAkagawa, 2020-12-12:
Are you using "USB Host Library SAMD" ?

Kuba0040, 2020-12-12:
Yes, I am.

YuuichiAkagawa, 2020-12-12:
This library is for [USB Host Shield 2.0 Library](https://github.com/felis/USB_Host_Shield_2.0). 
I can't support.

Isn't your problem the same as this?
https://github.com/gdsports/USB_Host_Library_SAMD/issues/8

closed 2020-12-12 by YuuichiAkagawa