YuuichiAkagawa/USBH_MIDI #65: Bidirectional_Converter example Clock Problem

jhsa, 2021-01-21:
Hi, I am experimenting with the Bidirectional_Converter example. 
I can receive "Clock" messages from a device connected to the USB Host, but it doesn't work if I try to send the Clock the other way, to the device connected to the host shield.
Help Please? Thank you.

YuuichiAkagawa, 2021-01-21:
The fix shown in #64 is incomplete. 
Try the following code.
```
    if (MIDI.read()) {
      msg[0] = MIDI.getType();
      switch (msg[0]) {
        case midi::ActiveSensing :
          break;
        case midi::SystemExclusive :
          //SysEx is handled by event.
          break;
        default :
          if( msg[0] < 0xf0 ){
              msg[0] |= MIDI.getChannel() -1;
          }
          msg[1] = MIDI.getData1();
          msg[2] = MIDI.getData2();
          Midi.SendData(msg, 0);
          break;
      }

```

jhsa, 2021-01-21:
Will this also fix the clock problem?
Thank you so much.

jhsa, 2021-01-21:
To answer my question, yes it is :) Thank you so much. All the other stuff like song position, continue, stop, etc, also seem to be working.
Would you update the example then?
Thank you so much. I will test this a little bit better, and if no other issue is found, I will close the issue.

jhsa, 2021-01-22:
Not sure I should open another issue, but if I disconnect the USB device from the host, it does not reconnect when I insert it back. Only when I reset the Arduino Pro-Mini, or power cycle everything.
Is this normal? Thanks.

YuuichiAkagawa, 2021-01-22:
> Would you update the example then?

I will update it at a later date.

Some controllers I own can be reconnected. However, Pro mini 8MHz feels unstable.
I cannot guarantee that any device will work without problems.

I'm not familiar with the guiter. Do you have a connection diagram?

jhsa, 2021-01-22:
well, today, the same device is reconnecting all the time. I don't understand this.. I did the report above and turned the PC off. Today just turned it on and it is re-connecting. Go figure that :)

My Pro-mini has a real crystal on it, not a resonator, so it should be more accurate. I replaced the fake m168 with a real m328. Will post the connections diagram I have shortly.

jhsa, 2021-01-22:
Ok, This is exactly how I have it connected. 

[image]

YuuichiAkagawa, 2021-01-24:
I connect the power supply without  a diode.
I'm sorry. What I want to know is the connection diagram of MIDI devices and micro controllers.
I don't know what you are trying to make.

jhsa, 2021-01-24:
Don't know what you mean then, The green numbers on the picture represent the Arduino pins those pins are connected to. I thought it was what you wanted to know?
The diode is there so the device can output 5V but cannot receive any voltage from other devices connected to it. Its a protection.
I am still having problems with devices not being recognized if I disconnect a device and connect the same one or another one, unless I reboot the arduino and shield.
I wonder if the circuit we make the cut is actually a detection to know when new devices are connected?
What i am trying to make at the moment is just a normal midi (DIN connectors) adapter to USB Host. Nothing complicated. Just an adapter.So, it is just the Arduino + USB shield + normal midi in/out.

YuuichiAkagawa, 2021-01-25:
> What i am trying to make at the moment is just a normal midi (DIN connectors) adapter to USB Host. Nothing complicated. Just an adapter.So, it is just the Arduino + USB shield + normal midi in/out.

Understood. Like this.
[image]
Reconnecting is not a MIDI driver issue so I have no solution.

jhsa, 2021-01-25:
Yeah, thank you. I thought this library was part of the other USB Host library. Therefore I was mixing things up. Well, thank you very much for your support and patience. and I do apologize again for my off topic posts.

closed 2021-01-25 by jhsa