YuuichiAkagawa/USBH_MIDI #68: BOSS GT-1 , Not working?

Nobu-Otsu, 2021-01-25:
BOSS　GT-1 Guitar Effects Processor とArudinoで通信を行いたいと思っています。
GT-1は、Id要求コマンドを受けることで、送受信が有効となる仕組みになっている様で、
Windowsとの接続テストでは以下で２番以降のコマンドを受け付けてます。
１．Win　→　GT-1　（ID要求）
　F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F0 7E 00 06 01 F7
２．Win　→　GT-1　（例：バッチ１番に切替）
　F0 41 00 00 00 00 30 12 00 01 00 00 00 00 7F F7

これをUSBH_MIDIで実装したいのですが、どうやら１番のコマンドが届かない様です、Winで１を送ったのちにUSBを差し替えて、USBH_MIDIで２を送ると問題なく動作するのです。
同一のArudino開発環境、ハードウェアでZOOMのMS-50G等でUSBH_MIDIを使った通信は問題なくできています。
GT-1がMIDIデバイスとして認識されていないこと等が原因でしょうか、ご教授ください。

MIDI Init
Addr:01
VID:0582 PID:01D6 #Conf:01

Conf:01 Int:00 Alt:00 EPs:00 IntCl:FF IntSubCl:FF
No MIDI Device

Conf:01 Int:01 Alt:00 EPs:00 IntCl:FF IntSubCl:02
No MIDI Device

Conf:01 Int:01 Alt:01 EPs:01 IntCl:FF IntSubCl:02
No MIDI Device
-EPAddr:0D bmAttr:05 MaxPktSz:68

Conf:01 Int:02 Alt:00 EPs:00 IntCl:FF IntSubCl:02
No MIDI Device

Conf:01 Int:02 Alt:01 EPs:01 IntCl:FF IntSubCl:02
No MIDI Device
-EPAddr:8E bmAttr:25 MaxPktSz:68

Conf:01 Int:03 Alt:00 EPs:02 IntCl:FF IntSubCl:03
No MIDI Device
-EPAddr:03 bmAttr:02 MaxPktSz:40
Endpoint descriptor:
 Length:	07
 Type:		05
 Address:	03
 Attributes:	02
 MaxPktSize:	0040
 Poll Intrv:	01
-EPAddr:84 bmAttr:02 MaxPktSz:40
Endpoint descriptor:
 Length:	07
 Type:		05
 Address:	84
 Attributes:	02
 MaxPktSize:	0040
 Poll Intrv:	00

Conf:01 Int:03 Alt:01 EPs:02 IntCl:FF IntSubCl:03
No MIDI Device
-EPAddr:03 bmAttr:03 MaxPktSz:40
-EPAddr:85 bmAttr:03 MaxPktSz:40

NumEP:03
MIDI not found. Attempts bulk device
Conf:01
EPin :84
EPout:03
Init done.

USB_desc
----------------------------
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	00
Device Protocol:	FF
Max.packet size:	40
Vendor  ID:		0582
Product ID:		01D6
Revision ID:		0000
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		00B0
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		00

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		00
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		02
Intf. Protocol:		02
Intf.string:		00
Unknown descriptor:
Length:		06
Type:		24
Contents:	F10100000904

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		FF
Intf. Subclass:		02
Intf. Protocol:		02
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01010001000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010204180144AC000705

Endpoint descriptor:
Endpoint address:	0D
Attr.:			05
Max.pkt size:		0168
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		02
Intf. Protocol:		01
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		01
Intf. Class:		FF
Intf. Subclass:		02
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		07
Type:		24
Contents:	01070001000B24
Unknown descriptor:
Length:		0B
Type:		24
Contents:	02010204180144AC000705

Endpoint descriptor:
Endpoint address:	8E
Attr.:			25
Max.pkt size:		0168
Polling interval:	01
Unknown descriptor:
Length:		07
Type:		25
Contents:	01000000000904

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00
Unknown descriptor:
Length:		06
Type:		24
Contents:	F10201010705

Endpoint descriptor:
Endpoint address:	03
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Interface descriptor:
Intf.number:		03
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00

Endpoint descriptor:
Endpoint address:	03
Attr.:			03
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	85
Attr.:			03
Max.pkt size:		0040
Polling interval:	01

Addr:1(0.0.1)

YuuichiAkagawa, 2021-01-25:
USBH_MIDIはMIDIクラスドライバなので、原則としてOSに専用のドライバをインストールしなくても認識できる機器でないと認識できません。認識できるのは
```
IntCl:01 IntSubCl:03
```
のものです。
特別なケースとして、MIDIクラスを持たない機器(IntCL:FF=ベンダ固有)でもバルクエンドポイントが該当する場合があるので、それをMIDIとみなして接続を試みます(手元の機器だとRolandのUM-ONE)。
BOSS GT-1も最終的に
```
MIDI not found. Attempts bulk device
Conf:01
EPin :84
EPout:03
Init done.
```
として接続を試みていますが、

> Winで１を送ったのちにUSBを差し替えて、USBH_MIDIで２を送ると問題なく動作するのです。

ということなので、恐らくWindowsのドライバが接続時に特別なUSBのコマンドを送信するとUSB MIDIが有効になるという仕組みの様です。
そのコマンドがわかれば単独で動作させることはできると思いますが...

Nobu-Otsu, 2021-01-25:
ありがとうございます。

不思議なのは、１も２の要求も返送があるのですが、Winに接続した後にUSBH_MIDIに接続替えをして、ID要求を行っても返送がないのです。２は要求を受けてた後にバッチ切替、返送もされます。

最初からUSBH_MIDIと接続して、1.ID要求、2.バッチ切替と送ると２のタイミングでGT-1に「bulk dataを受信した」的なメッセージがでます、これはWinで１を送らずに２を送った時も同様のメッセージがでます。
１のデータはUSBH_MIDIで送られてないのでは？と思った次第です。

USBアナライザー的なもので調べてみたいのですが、技術も知識も機材もなく残念です。

Winにつないだ後とつながない場合で、USB_desc、Midi_Initの動きをもう少し調べてみます。

YuuichiAkagawa, 2021-01-26:
そういうことですか。私の指摘は違いますね。接続自体はできているように思えます。
状況的にID要求が正しく送信できていないように見えますね。
ID要求に'F0'があってこれがちょっと怪しいので、仕事終わったら確認します。

Nobu-Otsu, 2021-01-26:
すばやいレスポンスに感謝です。’F0’怪しいですよね。
USBH_MIDI::SendSysEx()の以下の直前でbuf内容を確認したところでは、パケットの組み立ても0xF0のデータも正しく見えましたが、今夜再度テストしてダンプを添付してみます。

if( (rc = pUsb->outTransfer(bAddress, epInfo[epDataOutIndex].epAddr, wptr, buf)) != 0 ){

YuuichiAkagawa, 2021-01-26:
USBH_MIDIにはSysEx用データのエンコードやデコードの機能は実装されていません。そのまま送受信します。
F0-F7の間には正しいデータを指定する必要があります。つまり、SysExでも00-7Fまでのデータしか含めることができません。

Nobu-Otsu, 2021-01-26:
uint8_t USBH_MIDI::SendSysExにトレースを追加して、ID要求を送ってみました。
おっしゃる通りにF0も、そのまま送信しています。
MIDIデータは00-7Fまでのデータしか含められない規約なので、GT-1のWindowsのドライバがなんらかのエンコードをして「F0」を送っているのではないか。という理解でよろしいでしょうか。
--
ID要求
 F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F0 7E 00 06 01 F7
--       
uint8_t USBH_MIDI::SendSysEx（)
トレース追加
         if( wptr >= maxpkt || n == 0 ){ //Reach a maxPktSize or data end.
                        USBTRACE2(" wptr:\t", wptr);
/* add */    for(int i=0;i<wptr;i++)USBTRACE2(" buf:", buf[i]);
                            if( (rc = pUsb->outTransfer(bAddress, epInfo[epDataOutIndex].epAddr, wptr, buf)) != 0 ){
---
SendSysEx:
	 Length:	0014
 Total pktSize:	001C
 wptr:	1C
 buf:04
 buf:F0
 buf:41
 buf:00
 buf:04
 buf:00
 buf:00
 buf:00
 buf:04
 buf:30
 buf:12
 buf:7F
 buf:04
 buf:00
 buf:00
 buf:01
 buf:04
 buf:01
 buf:7F
 buf:F0
 buf:04
 buf:7E
 buf:00
 buf:06
 buf:06
 buf:01
 buf:F7
 buf:00
--------

YuuichiAkagawa, 2021-01-26:
https://www.vguitarforums.com/smf/index.php?topic=22578.msg164912#msg164912
これと同じかどうかわからないのですが、
```
Enter Editor mode     F0 41 00 00 00 00 3E 12 7F 00 00 01 01 7F F7
```
ではどうですか？

Nobu-Otsu, 2021-01-26:
Enter Editor mode（GT-1） F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F7

GT-1B用の3Eを30に変更することで、GT-1用になりました。
ありがとうございました。

私は、以下のツールのソースから、このID要求を知ったのです。
https://sourceforge.net/projects/fxfloorboard/files/GT-1FxFloorBoard/

globalVariables.h
idRequestString = "F0410000000030127F000001017FF07E000601F7";
idReplyPatern = "F07E000602413003000000000000F7";
--

正しく書くと以下が正解でMIDI-OXの結果とも一致します、
本当にありがとうございました。

Enter Editor mode
 F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F7
IDRequestString
 F0 7E 00 06 01 F7
IDReplyPatern
 F0 7E 00 06 02 41 30 03 00 00 00 00 00 00 F7

YuuichiAkagawa, 2021-01-26:
動いて良かったです。
ミスリード失礼しました。

Nobu-Otsu, 2021-01-27:
usbh_midiにとどまらず、GT-1Bの情報までも調べてくださりありがとうございました。

YuuichiAkagawa, 2021-01-27:
## Summary
### Q
I sent the following SysEx message, but GT-1 did not receive it.
```
F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F0 7E 00 06 01 F7
```
### A
USBH_MIDI :: SendSysEx() does not interpret the data and sends it as it is.
You need to separate manually.
```
F0 41 00 00 00 00 30 12 7F 00 00 01 01 7F F7
F0 7E 00 06 01 F7
```

closed 2021-01-27 by YuuichiAkagawa