YuuichiAkagawa/USBH_MIDI #69: Multiple Ports?

bolau, 2021-04-06:
Hi,
the hint to the cable number in SendData suggests that a USBH_MIDI device uses multiple virtual ports, i.e., shows up as multiple packages with 16 channels each. Is that really the case? If so, how many? Can you also distinguish in RecvData what cable the message came from?
Best, Boris

YuuichiAkagawa, 2021-04-06:
The Cable Number is value range from 0 to 15.
See page 16 of the USB MIDI specification.
https://usb.org/sites/default/files/midi10.pdf

You can receive it, but I can't test it because I don't own the corresponding device.
Like this
```
#include <usbh_midi.h>
USB Usb;
USBH_MIDI Midi(&Usb);

//USB-MIDI event packet
typedef union
{
  uint8_t buf[4];
  struct {
    uint8_t header;
    uint8_t byte1;
    uint8_t byte2;
    uint8_t byte3;
  } p;
} usbMidiEventPacket_t;

void setup() {
  Serial.begin(115200);
  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop()
{
  uint8_t msgbuf[4];
  usbMidiEventPacket_t *pkt = reinterpret_cast<usbMidiEventPacket_t*>(msgbuf);

  Usb.Task();
  if ( Midi ) {
    uint8_t size;
    do {
      size = Midi.RecvData(msgbuf, true);  //2nd paramter is enable raw packet mode
      if ( size > 0 ) {
        //Cable number
        uint8_t cn = ( pkt->p.header >> 4 ) & 0x0f;
        char strbuf[16];
        sprintf(strbuf, "%2d %02X %02X %02X",  cn, pkt->p.byte1, pkt->p.byte2, pkt->p.byte3);
        Serial.println(strbuf);
      }
    } while (size > 0);
    delay(100);
  }
}
```

YuuichiAkagawa, 2021-05-09:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2021-05-09 by YuuichiAkagawa

bolau, 2021-05-10:
Sorry for not getting back to you. Thanks for your reply, this will be very helpful. I also can't test it right now because I don't have an Arduino yet. I'm working with Teensys that have the USB host stuff integrated. But due to compatibility problems, I want to switch to Arduino with a USB host shield, and your library facilitates me porting my code over.
Best, Boris