YuuichiAkagawa/USBH_MIDI #70: USBH_MIDI : Failing to get device descriptor with some devices. (Using USBH_MIDI from gdsports)

soulwyx, 2021-04-08:
I'm using USBH_MIDI from gdsports, but I'm with a problem that maybe someone can helps me.

I can't get device descriptors at Init. (line 152, usbh_midi.cpp)

rcode = pUsb->getDevDescr( 0, 0, sizeof(USB_DEVICE_DESCRIPTOR), (uint8_t*)buf );
or
 rcode = pUsb->getDevDescr( 0, 0, 8, (uint8_t*)buf );
 
With some devices work fine, and with others doesn't.
(return 0 when successful ; 8 when error )

Looking deep, the problem come from:   (this is for samd21)
Line 206 from USB.cpp
rcode = dispatchPkt(tokSETUP, ep, nak_limit); // Dispatch packet

Any tips?

I can get device descriptor of any devices without using USBH_MIDI.
I can get device descriptor from my PenDrive, and return that isn't a MIDI Device.
I can't get device descriptor from my Mouse by example, returning the RCODE equal 8.
My Zoom G3 works fine. My Nux MG300 doesn't.

YuuichiAkagawa, 2021-04-08:
I don't know the SAMD version, but is this helpful?
https://github.com/YuuichiAkagawa/USBH_MIDI/issues/62

soulwyx, 2021-04-09:
Not for this case.
I having problem gettingDeviceDescriptor. I think that this will be a future problem if I break through this one first.

Any other tips?

Thank you so much.

soulwyx, 2021-04-09:
I got it!
I need to do many things that I found on Issues, so very thank you ! 
Let's explain:
Need to remove Release from USBH_MIDI.h
`//virtual uint32_t Release();`
and 
```
/*
uint32_t USBH_MIDI::Release()
{
        pUsb->GetAddressPool().FreeAddress(bAddress);
        bNumEP       = 1;               //must have to be reset to 1
        bAddress     = 0;
        bPollEnable  = false;
        readPtr      = 0;
        return 0;
}*/
```

Plus:

```
#define DESC_BUFF_SIZE        512
#define MIDI_EVENT_PACKET_SIZE 128
```

Without Release();
I have some fails when pluggin a device. And I always must restart my board to connect another device.

Do you have any suspicions to explain this? I know this is in SAMD21, but with your answer in the last post, I managed to make it happen!

YuuichiAkagawa, 2021-04-10:
Again, I don't know about the SAMD version.

The NUX MG-300 seems to have a large descriptor, so I think the following modifications are correct.
> #define DESC_BUFF_SIZE        512
> #define MIDI_EVENT_PACKET_SIZE 128

However I'm not sure if it has anything to do with getDevDescr () failing.

closed 2021-04-10 by YuuichiAkagawa