YuuichiAkagawa/USBH_MIDI #71: Midi Beat Clock

gcannar, 2021-04-09:
Hi there i'm trying to send midi beat clock to a moog synth from an Arduino Leonardo and a USB Host Shield 2.0 via usb cable.

I managed how to send notes but cannot send beat clock or start or stop messages. Can someone help plz? :)

I mean... this is working correctly and play note:

`Midi.SendData((uint8_t*)"\x90\x48\x7f");`

This not: (beat clock)

 `Midi.SendData((uint8_t*)"\x0F\xF8");`

Neither this: (start)

`Midi.SendData((uint8_t*)"\x0F\xFA");`

Any suggestions? Thanks!

YuuichiAkagawa, 2021-04-10:
Try the following:
```
   Midi.SendData((uint8_t*)"\xf8");
   Midi.SendData((uint8_t*)"\xfa");
   Midi.SendData((uint8_t*)"\xfc");
```

YuuichiAkagawa, 2021-05-09:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2021-05-09 by YuuichiAkagawa