YuuichiAkagawa/USBH_MIDI #72: v0.6.0

YuuichiAkagawa, 2021-04-18:
Developing version 0.6.0 on the "develop" branch.
https://github.com/YuuichiAkagawa/USBH_MIDI/tree/develop

- [x] Change configuration descriptor parser
- [x] Fixed an issue when the endpoint size exceeded 64 bytes
- [x] Add OnInit() callback
- [x] Add a predefined macro "USBH_MIDI_VERSION"
- [ ] Refactoring

###  Change configuration descriptor parser
Changed the parser of the configuration descriptor to a confdescparser.h based implementation.
It now works fine with large descriptors.
- #48 
- #62

### Fixed an issue when the endpoint size exceeded 64 bytes
USB Host Shield supports up to 64 bytes.
If the packet size of the endpoint descriptor is larger than 64, limit it to 64.
- #62
@DAsaroth Can you test this?

### Add OnInit() callback
Register a user function to call when the controller is successfully initialized.

### Add a predefined macro "USBH_MIDI_VERSION"
A macro has been added so that the version of the library can be identified.

### Refactoring
In progress.

DAsaroth, 2021-04-19:
@YuuichiAkagawa, sure, in the next few days I'll try it. I'll let you know.  
Thank you

DAsaroth, 2021-04-21:
@YuuichiAkagawa , Hi, I have updated my project with the new libraries, everything seems to work.
Thank you

YuuichiAkagawa, 2021-04-21:
@DAsaroth  Thank you for your cooperation.

DAsaroth, 2021-04-29:
You are welcome, thanks to you for the support and the great library

closed 2021-05-09 by YuuichiAkagawa