YuuichiAkagawa/USBH_MIDI #73: Keystation 49 MK3 recognized but no MIDI events received

Nephiel, 2021-04-25:
This MIDI keyboard model is class compliant and seems to be recognized, but no MIDI events are received at all.

Here is the **USB_MIDI_dump** with `#define DEBUG_USB_HOST` :
```
MIDI Init
Addr:01
VID:0763 PID:3110 #Conf:01

Conf:01 Int:00 Alt:00 EPs:00 IntCl:01 IntSubCl:01
No MIDI Device

Conf:01 Int:01 Alt:00 EPs:02 IntCl:01 IntSubCl:03
MIDI Device
-EPAddr:81 bmAttr:02 MaxPktSz:40
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	81
 Attributes:	02
 MaxPktSize:	0040
 Poll Intrv:	00
-EPAddr:02 bmAttr:02 MaxPktSz:40
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	02
 Attributes:	02
 MaxPktSize:	0040
 Poll Intrv:	00

NumEP:03
Conf:01
EPin :81
EPout:02
Init done.
VID:0763, PID:3110
```
Tried all of the keys, wheels, fader and buttons, but nothing appears on the dump after this point.

**USBH_MIDI_desc** output:
```
VID:0763, PID:3110
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	08
Vendor  ID:		0763
Product ID:		3110
Revision ID:		0105
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0085
Num.intf:		02
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		32
[... 274 lines cut ...]
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 6
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         2
          baAssocJackID( 0)       1
          baAssocJackID( 1)       3
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

Let me know if I can help get this keyboard model supported.

YuuichiAkagawa, 2021-04-26:
Do you have enough power ?
Can you try connecting via a Powered USB hub?

Please check inTransfer() return value.

https://github.com/YuuichiAkagawa/USBH_MIDI/blob/808b5a1e65496cbd4daab4bebdcb841640c6e682/usbh_midi.cpp#L377-L379
↓ insert #ifdef - #endif
```
        uint8_t  r = pUsb->inTransfer(bAddress, epInfo[epDataInIndex].epAddr, bytes_rcvd, dataptr);
#ifdef DEBUG_USB_HOST
        if( r )
                USBTRACE2("inTransfer():", r);
#endif
        if( *bytes_rcvd < (MIDI_EVENT_PACKET_SIZE-4)){
```

Nephiel, 2021-04-26:
Here's the dump with check in place:

```
[...]
NumEP:03
Conf:01
EPin :81
EPout:02
Init done.
VID:0763, PID:3110
inTransfer():04
inTransfer():04
inTransfer():04
inTransfer():04
inTransfer():04
inTransfer():04

```
Steady stream of `InTransfer():

YuuichiAkagawa, 2021-04-26:
Hmmm
The Keystation 49 MK3 responds to NAK, so USB communication is possible. But I don't understand why it doesn't send MIDI events.

YuuichiAkagawa, 2022-02-17:
Try commenting out the following line in Usb.cpp.
https://github.com/felis/USB_Host_Shield_2.0/blob/9184254b87b2c65378ba83d653457a48c01d5396/Usb.cpp#L241-L246

Like this.
```C++
                rcode = dispatchPkt(tokIN, pep->epAddr, nak_limit); //IN packet to EP-'endpoint'. Function takes care of NAKS.
//                if(rcode == hrTOGERR) {
//                        // yes, we flip it wrong here so that next time it is actually correct!
//                        pep->bmRcvToggle = (regRd(rHRSL) & bmRCVTOGRD) ? 0 : 1;
//                        regWr(rHCTL, (pep->bmRcvToggle) ? bmRCVTOG1 : bmRCVTOG0); //set toggle value
//                        continue;
//                }
                if(rcode) {
```

doctea, 2022-02-17:
> Try commenting out the following line in Usb.cpp. https://github.com/felis/USB_Host_Shield_2.0/blob/9184254b87b2c65378ba83d653457a48c01d5396/Usb.cpp#L241-L246

Success!  Sending and receiving MIDI note on/offs now seems to work!  Thank you! :D

Are there any drawbacks to this modification -- will this have a negative impact on the of other devices?  (I've tested it with an Akai APCmini, and that seems to work, but maybe there is a case where it causes a problem?)

I note that although noteon/noteoff messages seems to work fine in both directions, that using MIDI.sendClock() to send clock to the Beatstep doesn't seem to work -- is this perhaps handled in a different function that needs to be similarly modified?  I have also tested with an Arturia Keystep, and that DOES receive the MIDI clock OK.

(I also find that neither the Arturia Beatstep or the Keystep are detected as MIDI devices if connected through a USB hub - I'll open a separate issue for this.)

doctea, 2022-02-18:
> @doctea Sorry, I made a mistake in one place, please try again. Only two lines need to be commented out.

Thanks -- I've changed my copy of the library as you suggested here.  It doesn't seem to make any difference that I can see, though.  The Beatstep still doesn't respond to clock - was this meant to solve this problem?  I've been through the settings in the Beatstep control app, hoping to find something like a 'MIDI clock mode' option, but had no such luck, so I don't know what else to try here..
 
> This problem is on the USB_Host_Shield_2.0 side, not the MIDI driver. [felis/USB_Host_Shield_2.0#438](https://github.com/felis/USB_Host_Shield_2.0/pull/438)

Would it be better to solve this problem by applying that pull request to my copy of the USB_Host_Shield_2.0 library rather than commenting out the two lines you suggest?

doctea, 2022-02-19:
> @doctea I thought that the cause of not working in BeatStep might be the toggle problem and suggested commenting it out. During my research, I found a pull request for USB_Host_Shield_2.0 and introduced it to you. If this pull request fix solves the problem, then it should be adopted.

Many thanks for your help, I understand now!  Yes it seems to solve the problem and should be adopted :)

YuuichiAkagawa, 2023-03-21:
@Nephiel 
OK, I purchased Keystation for verification.
I have confirmed that it works in two cases: with an external power supply connected to the Aruduino and through a powered USB hub.

> Also tested power draw with a meter, Arduino Pro Micro + USB Host board + Keystation MK3 only draw about 60 mA.

Unfortunately, that is current consumption under stable conditions. This keyboard draws a large current when it is powered on.
[image]

I will say it again. You must connect via a powered USB hub.

I've wasted my money, you know. :sob:

closed 2023-03-21 by YuuichiAkagawa