YuuichiAkagawa/USBH_MIDI #74: Help to implement control on Mooer Ge200.

soulwyx, 2021-09-15:
Since I'm running with SAMD21 Host library, I can't run your example.
I run this with Thesycon USB Descriptor Dumper. (in the end of issue)
What I know at this moment:

**This go to preset 3.**
```
uint8_t preset3[64]={	0x08,0xAA,0x55,0x02,0x00,0xA6,0x02,0x85,0x0D,0x00,0xE0,0x0B,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
```

**This go to preset 4.**
```
uint8_t preset4[64]={	0x08,0xAA,0x55,0x02,0x00,0xA6,0x03,0x95,0x2C,0x00,0xE0,0x0B,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
```

Using HID Send (https://docs.hhdsoftware.com/dms/advanced-features/usb-monitoring/hid-send.html), I can control Mooer Ge200 using the messages above.

My questions is how to  properly initiate the device and send this message using USBHost library.
Could someone give me any tips or directions?

```
#include <Arduino.h>
#include <usb.h>
#include <usbh_midi.h>

USBHost usb;
USBH_MIDI  midiHostUsb(&usb);

byte i;
uint8_t preset3[64]={	0x08,0xAA,0x55,0x02,0x00,0xA6,0x02,0x85,0x0D,0x00,0xE0,0x0B,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
	
	uint8_t preset4[64]={	0x08,0xAA,0x55,0x02,0x00,0xA6,0x03,0x95,0x2C,0x00,0xE0,0x0B,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
		0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
	0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};

void setup() {
	startDisplay();
	usb.Init();
		
	while(1){
		usb.Task();
		delay(100);
		if(usb.getUsbTaskState() == USB_STATE_RUNNING){
                        midiHostUsb.SendMooerData(preset4);
			delay(1000);
		}
	}
}
```

`uint8_t USBH_MIDI::SendMooerData(uint8_t *dataptr){
	return pUsb->outTransfer(bAddress, 0x81, 0x40, dataptr);
}`

With my code I get rcode=219. (USB_ERROR_EP_NOT_FOUND_IN_TBL)
What I'm missing?

Information for device GE200 (VID=0x0483 PID=0x5703):

*** ERROR: Descriptor has errors! ***

[... 299 lines cut ...]
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "MOOER"
0x02   0x0409  "GE200"
0x03   0x0409  "00000000001A"

------------------------------

Connection path for device: 
Controlador Host Compatível com USB xHCI
Root Hub
GE200 (VID=0x0483 PID=0x5703) Port: 2

Running on: Windows 10 or greater (Build Version 19042)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

YuuichiAkagawa, 2021-09-16:
The descriptor does not appear to have a MIDI interface.
Does this device support USB MIDI?

soulwyx, 2021-09-16:
```
uint8_t USBH_MIDI::SendMooerData(uint8_t *dataptr){
	
	
	epInfo[epDataOutIndex].maxPktSize=0x40;
	epInfo[epDataOutIndex].epAddr=0x02;
	epInfo[epDataOutIndex].bmAttribs=0x03;
	

	pUsb->setEpInfoEntry(bAddress, 3, epInfo);
	
	return pUsb->outTransfer(bAddress,epInfo[epDataOutIndex].epAddr, epInfo[epDataOutIndex].maxPktSize, dataptr);
}
```

With this implementation now I'm getting Rcode=8 (USB_ERRORTIMEOUT) from dispatchPkt
What I'm still missing?

soulwyx, 2021-09-16:
> The descriptor does not appear to have a MIDI interface.
> Does this device support USB MIDI?

No, this device don't. I'm trying to force a message.
I can control on my computer sending the message on my computer.

YuuichiAkagawa, 2021-09-16:
This library will not work without a USB MIDI device. 
Therefore, no matter how hard you try, you will not be able to send messages.

soulwyx, 2021-09-16:
> This library will not work without a USB MIDI device.
> Therefore, no matter how hard you try, you will not be able to send messages.

Even if I try to create a new function inside?
Can you suggest me a library to make this happen?

YuuichiAkagawa, 2021-09-16:
Your device is not MIDI, so modifying this library will not help.
We cannot advise on device-specific matters.

closed 2022-03-13 by YuuichiAkagawa