YuuichiAkagawa/USBH_MIDI #77: USB-MIDIインターフェースからうまくMIDI信号を出せません

tacotsubo, 2023-08-05:
Xでコメントをいただきありがとうございます。
プログラミングは初心者です。

M5StackにUSBアダプタを繋いでいます。
USB端子にはハブ、その先に2台のコルグの入力デバイス、ローランドのUSB-MIDIインターフェースを繋いでいます。
入力デバイスからインターフェースに信号は出るようなのですが、SC-8850という音源の画面には
「MIDI Off Line」というメッセージが出るだけで音が出てきません。
マニュアルによるとこのメッセージはMIDIケーブルの断線や信号途絶の意味があるようです。
SendData関数の使い方に間違いがあるのかと思っています。

以下に組んだソースをお示ししますのでご意見をいただければ幸いです。

なお精神疾患を持っていますので健常者の方とのやりとりが難しい面もあるかもしれません。
そうお感じになったらお伝えいただけますでしょうか。

// include the library code:
#include <usbh_midi.h>
#include <M5Stack.h>
#include <usbhub.h>

USB Usb;
USBHub Hub(&Usb);
USBH_MIDI  Midi1(&Usb);
USBH_MIDI  Midi2(&Usb);
USBH_MIDI  Midi3(&Usb);
uint8_t bufMidi[64];
void MIDI_poll();

uint16_t pid, vid;

void setup() {
  M5.begin();
  vid = pid = 0; 
  Serial.begin(115200);
  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
}

void loop() {
  Usb.Task();
  if ( Midi1 ) {
  //uint32_t t1 = (uint32_t)micros();
      MIDI_poll1();
  }
  //uint32_t t1 = (uint32_t)micros();
  if ( Midi2 ) {
    MIDI_poll2();
  }
  //uint32_t t1 = (uint32_t)micros();
  if ( Midi3 ) {
    MIDI_poll3();
  } 
}

// Poll USB MIDI Controler and send to serial MIDI
void MIDI_poll1()
{
  uint16_t  rcvd;
  if (Midi1.RecvData( &rcvd,  bufMidi) == 0) {
     Midi3.SendData(bufMidi);
  }
}

void MIDI_poll2()
{
  uint16_t  rcvd;
  if (Midi2.RecvData( &rcvd,  bufMidi) == 0) {
    Midi3.SendData(bufMidi);
  }
}
void MIDI_poll3()
{
  if(bufMidi[1]!=0x00){
    Midi3.SendData(bufMidi);
  }
}

YuuichiAkagawa, 2023-08-05:
USBH_MIDIならびに、依存している[USB Host Shield 2.0 Libraryでは、どのインスタンスにどのUSB機器が割り当てられているかを知る仕組みが用意されていません。
tacotsuboさんのコードでは、Midi3を出力用のインスタンスと決め打ちしていますが、その保証はありません。
ご自身のコーディングで、それぞれどのインスタンスがどのUSB機器なのかを管理する必要があります。

```RecvData(uint16_t *bytes_rcvd, uint8_t *dataptr);```で受信したデータはRawデータかつUSB MIDI機器によっては有効でないデータを含んだデータ長を返します。
例えば、1メッセージだけしか送られてきていないのに64バイト受信する場合があります。そのため、1メッセージずつ処理する必要があります。

今回tacotsuboさんがやろうとしていることに近しい例を以下に示します。
https://gist.github.com/YuuichiAkagawa/0841a1f1fee67c9726a86e688851f56a

USB HUBを経由してnanoKEY2キーボードとRoland UM-ONE mk2を接続し、MIDI接続したVolca KeysをnanoKEY2で演奏できることを確認済みです。(Arduino UNOで確認しています)
setupmidi()関数内でUSB MIDI機器を識別しています。予め、ご自身が利用するUSB MIDI機器のベンダIDとプロダクトIDを調べておく必要があります。IDが不明な場合はUSB_MIDI_desc.inoをご利用ください。
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USB_MIDI_desc/USB_MIDI_desc.ino

最後に「MIDI Off Line」ですが、SC-8850の仕様を知らないため憶測となりますが、[アクティブセンシング](https://www.g200kg.com/jp/docs/dic/activesensing.html)メッセージを送信し続けなければならないのではないでしょうか。
http://faq-jp.roland.com/faq/show/75936?category_id=8554&site_domain=default
こちらはご自身で送信する処理をコーディングする必要があります。
(ここは本ライブラリの範囲外の部分なので、ご自身で調査願います)

tacotsubo, 2023-08-05:
詳細なご説明ありがとうございます。
またお世話になるかと思いますがその折はどうぞよろしくお願いします。

tacotsubo, 2023-08-15:
先日ご提案いただいた2デバイスを両方とも持っていたのでM5Stackで試してみました。
ArduinoのUSBホストシールドを持っていないのが理由です。
それぞれのデバイスの電源ランプは点くのですが、キーボードからの信号がMIDIインターフェースに送られていないようです。
なおCubaseProで双方のデバイスが使えることは確認しています。
どういった点を確認すればよいかご助言いただければ幸いです。

YuuichiAkagawa, 2023-08-15:
こちらの環境では問題なく動作しますので、本ライブラリやUSB Host Shield Libraryの問題では無いと考えます。
当方、M5StackやESP32には詳しくないのでマイコン自体の使い方についてはサポートできません。

---
### テスト環境
#### Micro controller
- M5Stack BASIC Core v2.4
- USB Module with MAX3421E

#### MIDI controller
- Roland UM-ONE mk2 (sw: COMP)
- KORG nanoKEY2

#### MIDI Instrument
- KORG volca keys (MIDI CH:1)

#### Build environment
Arduino 1.8.19を使用。OSはUbuntu 22.04。
- Board ManagerからM5Stack 2.0.7をインストール
- Library ManagerからUSB Host Shiled Library 2.0 Version 1.6.2 をインストール
- Boardに M5Stack-Core-ESP を指定
- [USBH_MIDI_2Dev_Kbd_Inst.ino](https://gist.github.com/YuuichiAkagawa/0841a1f1fee67c9726a86e688851f56a)をビルド＆書き込み

以上で、問題なく動作しています。

[image]

---
USB Host Shieldを利用する際の 一般的な確認方法
1. 電力は足りているのか？
マイコンの電源をPCのUSBから取っている場合は電力が不足している場合があります。
- Arduinoであれば、DCジャックから外部電源を供給するか、モバイルバッテリーから電力を供給してください
- 外部電源の接続できるUSBハブを使用してください。

2. そもそもUSB機器として認識できているのか？
 Examples>USB Host Shield Library 2.0>USB_Descを書き込み、USB ModuleにUSBハブを使わずに、一つずつUM-ONE や nanoKEY2を接続して Arduino IDEのシリアルモニタに情報が表示されるかを確認。(再実行にはマイコンの再起動が必要)

3. USB MIDI機器として認識しているのか？
Examples>USB Host Shield Library 2.0>USBH_MIDI>USBH_MIDI_dumpを書き込み、USB ModuleにUSBハブを使わずに、 nanoKEY2を接続し、鍵盤を押すとMIDIメッセージ内容が表示されることを確認。
※UM-OMEの確認はAudio I/FのMIDI OutにUM-ONEのConnect to MIDI OUTを接続し、CubaseからMIDI noteなどを送信してください。

4. その他
- 正しく動作している場合は、鍵盤を押すとUM-ONEのOUT LEDが点滅するはずです。
- USBのVID/PIDがお手元の機器と合っているか確認してください。
https://gist.github.com/YuuichiAkagawa/0841a1f1fee67c9726a86e688851f56a#file-usbh_midi_2dev_kbd_inst-ino-L30
https://gist.github.com/YuuichiAkagawa/0841a1f1fee67c9726a86e688851f56a#file-usbh_midi_2dev_kbd_inst-ino-L36
- MIDIのチャンネルが合っているか確認してください。サンプルはCH1に送信しています。

正しく動作しない場合
- 上記の確認項目はいずれも私の環境で問題なく動作しているのでUSB Host Shiled Library には問題ありません
- ハードウェアの問題であればM5Stackの販売店にお問い合わせください
- Arduino IDEのセットアップの問題であれば、M5Stack社にお問い合わせください
- 本ライブラリの使用にはある程度のUSBに関する知識が必要です。適切な資料を参照ください。
- コード例が動作しない場合は、ご自身で動くように修正してください

tacotsubo, 2023-08-15:
詳細なご説明ありがとうございます。おそらく当方の知識不足が原因です。
場合によってはUSBデバイスに拘らず別の手段も考えてみたいと思います。

YuuichiAkagawa, 2024-01-09:
Closing this issue due to lack of feedback. Feel free to reopen the issue again if needed.

closed 2024-01-09 by YuuichiAkagawa