YuuichiAkagawa/USBH_MIDI #79: Novation Launchpad Mk1 has problem.

rediff666, 2024-12-05:
I have a problem with My Launchpad Mk1 (its no recognized) , this library is not compatible with that controller? 
This is the description:

01
--
String Descriptors:
Manufacturer: Novation DMS Ltd
Product: Launchpad

Device descriptor:
Descriptor Length: 12
Descriptor type: 01
USB version: 0100
Device class: FF
Device Subclass: 00
Device Protocol: FF
Max.packet size: 08
Vendor  ID: 1235
Product ID: 000E
Revision ID: 0002
Mfg.string index: 01
Prod.string index: 02
Serial number index: 00
Number of conf.: 01

Configuration descriptor:
Total length: 0020
Num.intf: 01
Conf.value: 01
Conf.string: 00
Attr.: 80
Max.pwr: D7

Interface descriptor:
Intf.number: 00
Alt.: 00
Endpoints: 02
Intf. Class: FF
Intf. Subclass: 00
Intf. Protocol: 00
Intf.string: 00

Endpoint descriptor:
Endpoint address: 81
Attr.: 03(Interrupt)
Max.pkt size: 0008
Polling interval: 0A

Endpoint descriptor:
Endpoint address: 02
Attr.: 03(Interrupt)
Max.pkt size: 0008
Polling interval: 0A

Addr:1(0.0.1)

YuuichiAkagawa, 2024-12-05:
Try changing the following line:
https://github.com/YuuichiAkagawa/USBH_MIDI/blob/bd8f4b5217303a0dc23dde7ed48d94ec1a07687d/usbh_midi.cpp#L299
to
```C++
if( pid == 0x0e || pid == 0x20 || pid == 0x36 || pid == 0x51 || pid == 0x69 ) {
```

rediff666, 2024-12-06:
Yes I did and it still doesn't work. I tried changing the cable and it didn't work, I thought my usb host shield might be failing but when I connect it with a mouse or keyboard they work without problems. I changed the version of the Arduino IDE and it didn't work. I even tried on different computers one with Windows and another with Linux and it didn't work on any of them. I made a simple program to show you what I get in the Serial Monitor of the Arduino IDE. The output is the following:
USB Host Shield initialized
No MIDI device detected

The Program here:
#include <usbh_midi.h>
#include <usbhub.h>

// Initialization of USB Host and MIDI
USB Usb;
USBH_MIDI Midi(&Usb);

void setup() {
  Serial.begin(9600);

  // Initialize USB
  if (Usb.Init() == -1) {
    Serial.println("Error initializing USB Host Shield");
    while (1); // Halt if it fails
  }
  Serial.println("USB Host Shield initialized");

  // Wait a few milliseconds to ensure connection
  delay(2000);

  // Check if the MIDI device is connected
  if (Midi) {
    Serial.println("MIDI device detected.");
  } else {
    Serial.println("No MIDI device detected.");
  }
}

void loop() {
  Usb.Task(); // Update USB Host Shield
}

YuuichiAkagawa, 2024-12-06:
My Launchpad mini is also unstable, but it can sometimes be recognized when using a USB HUB.